ArduPilot/ardupilot #14695: copter 3.6.11 (pixhawk fmuv2) automission zigzag behavior

roque-canales, 2020-06-26:
Hello,

On this log we have strange zigzag behavior during automission....

GPS works perfectly, also there is other flights with the same behavior.

It's seems to appear since we have activated smartrtl...

Can you help me to find from where came this behavior?

[32 04-06-2020 15-34-56.zip](https://github.com/ArduPilot/ardupilot/files/4837664/32.04-06-2020.15-34-56.zip)

peterbarker, 2020-07-02:
Are you able to replicate this using the current Copter stable release?

muramura, 2020-07-11:
@roque-canales san, @peterbarker san.  I recommend updating to "3.6.12" before moving from "3.6.11" to "4.x.x". The "I2C Storm" bug is alive and well in 3.6.11.
The I2C Storm bug is alive and well in 3.6.11.
There are many FCs that connect the compass to I2C.
I think updating to "4.x.x" will make it harder to identify the cause of the zigzagging flight.

peterbarker, 2020-11-24:
@roque-canales if problems persist, please move to 4.0.x (or the latest stable) and test again.  If it is still a problem, reopen this issue.

closed 2020-11-24 by peterbarker