ArduPilot/ardupilot #14723: Internal Error after bootloader upgrade

rmackay9, 2020-07-01:
After performing the bootloader upgrade on a CubeOrange, "Internal Errors 0x9000 I:1009, iomcu_reset ,main_loop_stk" was displayed on the HUD.  I think this is probably innocuous but it's still a bit worrying for the user to see these types of errors when everything is working ok

2020-07-01 9:49:53 AM : PreArm: Internal errors 0x9000 l:1009 ,iomcu_reset,main_loop_stk
2020-07-01 9:49:53 AM : PreArm: IOMCU is unhealthy
2020-07-01 9:48:44 AM : Flash OK
2020-07-01 9:48:44 AM : Erasing
2020-07-01 9:48:32 AM : Frame: QUAD
2020-07-01 9:48:32 AM : RCOut: PWM:1-12
2020-07-01 9:48:32 AM : CubeOrange 001D0023 34385113 39373231
2020-07-01 9:48:32 AM : ChibiOS: 331fe75d

peterbarker, 2020-07-01:
Two separate issues
 - raising an internal error when the MCU resets when we could reasonably expect it to
   - Fix: https://github.com/ArduPilot/ardupilot/pull/14724
 - main loop stuck
   - this is odd as the monitor thread shouldn't be active until `system_initialised()` is called and that should happen after the IOMCU has been flashed.

peterbarker, 2020-07-01:
There is a race in the monitor thread.

The watchdog is patted just before we set the scheduler as initialised.

We delay 100ms while waiting for initialisation in the top of the monitor thread.

Then we delay 100ms as the first thing in the monitor thread loop.

That may mean we meet-or-exceed the 200ms timeout for deciding we have a main loop hangup.

I can't see why this would trigger on iomcu flash yet.

peterbarker, 2020-07-01:
`_initialised` is accessed across threads but isn't marked volatile (`hal_initialised` is)

peterbarker, 2020-07-01:
Fixes here: https://github.com/ArduPilot/ardupilot/pull/14728

Flamer999, 2020-10-14:
В продолжение темы хочу сообществ, что вчера разбил свой окто при падении с 45м. В полёте возникла ошибка Internal error iomcu_reset, что вызвало отключение Дрона, переворот по роллу и его падение. Чуть позже скину видео и логи

peterbarker, 2024-07-31:
We merged a fix for this.  Years ago.

@Flamer999 's problem - while unfortunate - would not have been related to this issue.

closed 2024-07-31 by peterbarker