ArduPilot/ardupilot #14738: Copter/Rover: ensure fence checks are not firing when !position_ok

rmackay9, 2020-07-03:
When Copter's position_ok() function is returning false we cannot trust the position regardless of what the AHRS says.  We should disable fence checks (and avoidance based on these checks) in these situations.

When we make this change we need to be sure that it works for both GPS and non-GPS position estimation

muramura, 2020-07-05:
I think the challenge is that this is a bug.
If you don't have this decision, does the failsafe trigger before you hit FENCE?
If not, is the next fence (+20m) created over the fence?
To check this in SITL, can you use the position_ok method and return false to see the effect?

muramura, 2020-07-05:
@rmackay9 san.
Will this action improve the situation?

https://github.com/ArduPilot/ardupilot/compare/master...muramura:AP_%2314738_bug_fix?expand=1

rmackay9, 2020-07-05:
@muramura, thanks for this.  I think this is a decent solution so a PR would be good, txs.

A few small changes:

- let's rename the check function's argument to "position_ok"
- no need for the extra check() that takes no arguments.  Sub and Rover also have equivalent functions. In Sub it's also "position_ok()", in Rover it is "ekf_position_ok()".

muramura, 2020-07-06:
@rmackay9 san... Thanks for the review.
I will reflect and PR the comments.