ArduPilot/ardupilot #14757: Copter: AltHold::init() should check EKF altitude is healthy

rmackay9, 2020-07-06:
AltHold, Sport and possibly all other modes except Stabilize and Acro (i.e. those with manual throttle) should check that the EKF has a good altitude estimate before switching into the mode when armed (i.e. add to the mode's init() function).

We may also need to add a pre-arm/arming check that we have a good altitude estimate and it should not be possible to bypass this check by setting ARMING_CHECK = 0.  There is a similar existing check that ensures that users cannot arm in autonomous modes without a good position estimate so this may be a good reference  to base the change on.

Note that it should always be possible to get into any mode while disarmed.

When testing whether the fix works or not, it should be possible to disable the barometer in SITL (I hope we have a SIM_BARO_FAIL or similar parameter) and then confirm that the vehicle cannot enter AltHold mode when armed and that the vehicle cannot arm in AltHold.

peterbarker, 2020-07-06:
A `required_altitude_estimate`, in the same vein as `requires_GPS` would probably be appropriate, then another block underneath the existing check in mode.cpp.

https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/mode.cpp#L246

One interesting case to be handled: I think guided mode can sometimes require an altitude estimate and sometimes not.

Also need to decide if the check returns false if home is not set.

rmackay9, 2020-07-06:
@peterbarker, 

Just a few points:

- I think we can just use the existing has_manual_throttle() method instead of creating a new one
- Guided mode always requires a valid altitude estimate
- home is not required

rmackay9, 2020-08-20:
The fix has been merged to master so closing

closed 2020-08-20 by rmackay9