ArduPilot/ardupilot #14760: Keep returning error “prearm check failed - check messages”

2020614, 2020-07-06:
## Bug report
Keep returning error “prearm check failed - check message“
**Issue details**
[image]
[image]

_Please describe the problem_
Keep returning error “prearm check failed - check message“，and can't switch auto mode（returned messages “PreArm:Need postition Estimate”）

**Version**
ArduRover V4.1.0-dev

**Platform**
- [ ] Rover

**Hardware type**
CUAV V5+

**Logs**
[https://1drv.ms/u/s!AsRqGtahBmzMj2zD4l-ZuY053DIK?e=Af2IpR](url)

amilcarlucas, 2020-07-06:
Auto mode is only available if position estimate (GPS has lock) is available.
So once you fix that, the error will go away. This is not a software bug.

closed 2020-07-06 by amilcarlucas

2020614, 2020-07-07:
> 
> 
> Auto mode is only available if position estimate (GPS has lock) is available.
> So once you fix that, the error will go away. This is not a software bug.

Hello, I conducted the test again, the results show that the above situation will also occur in the case of GPS lock。

[image]

peterbarker, 2020-07-11:
@amilcarlucas 's is a somewhat simplified version of reality.

What the vehicle needs is to know where it is; usually this is a GPS and sufficiently good other sensors.

In this case you do appear to have lots of satellites.  But you can also see that the vehicle is still unhappy with its position - "EKF failsafe" is a reasonable demonstration of that.

Often this is caused by a bad GPS fix, and often by poor sensor calibration.

That "Vibe" there may be an indication that the accelerometers are being swamped by the amount of vibration being dumped into your autopilot.  Is this running on a machine with an Internal Combustion Engine?

2020614, 2020-07-12:
> @amilcarlucas 's is a somewhat simplified version of reality.
>
> What the vehicle needs is to know where it is; usually this is a GPS and sufficiently good other sensors.
>
> In this case you do appear [...]

Thanks, he runs on a vehicle equipped with an internal combustion engine. The problem has been solved, it is caused by compass interference. The cause of interference may be vibration or surrounding magnetic field. and it is in use now “aka Moving Baseline”。

rmackay9, 2020-07-12:
Just a small correct, the "Vibe" message on MP's HUD is always there.  It will turn orange or red if the vibration levels are too high.