ArduPilot/ardupilot #14763: AC_AttitudeControl: Duplicate parameter documentation is generated

WickedShell, 2020-07-06:
## Bug report

**Issue details**

There are duplicate docs for the `ATC_` parameters generated due to both [AC_AttitudeControl_Heli.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp#L17](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl_Multi.cpp#L10) defining the same parameters. The net effect is the generated documentation either includes a duplicate parameter key, which means a GCS will sometimes select the wrong doc string for a vehicle, or in the case of the EDN documentation generator an invalid doc string is built.

The only solutions I can see is to either:
1. Unify on a single set of ranges/increments for these descriptions
2. We need to add a `Heli` tag to live alongside the `Copter` `Plane` `Rover` tags so that we have an entire extra parameter file for heli. The downside to this solution of course is all the GCS's need to update to reference this file, but they are almost gurantted to be displaying the wrong information sometimes at the moment.

**Version**
4.x Stable and Master

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2020-07-07:
@meee1, It sounds like MP could be impacted by this.  Do you think you could modify MP to pull parameter files for tradheli from a new file?  @peterbarker knows where this new file will be.

WickedShell, 2020-07-07:
@DonLakeFlyer Would adding a separate file for TradHeli on QGC be an issue?

DonLakeFlyer, 2020-07-07:
Not really an issue, requires code change in QGC to support. Fairly simple.

Better to also to move to new `COMPONENT_INFORMATION` parameter meta data support which is the thing which will standardize all of this. At least from a GCS standpoint.