ArduPilot/ardupilot #14769: ArduPlane-beta: cached parameter count errors

peterbarker, 2020-07-07:
## Bug report

**Issue details**

AP_Param's cached parameter count seems to get stale.

The situation can be reproduced on the bench.

Run the autopilot out of memory:
```
param set SCR_ENABLE 1
param set GRIP_ENABLE 1
param set CAN_P1_DRIVER 1
param set CAN_P2_DRIVER 1
param set ADSB_ENABLE 1
param set AVD_ENABLE 1
param set BTN_ENABLE 1
param set CHUTE_ENABLED 1
reboot
```

On reboot, you should get messages to the console about EKF2 being unable to allocate.

param fetch - you should note the parameters don't all download; it stalls short of its target.

```
param set GRIP_ENABLE 1
```

You should note that the parameter count in the MAVProxy console is instantly correct.
```
param fetch
```

You should note the parameters all successfully download.  enable/disable of a parameter tree invalidates the cached count.

There's one obvious race condition in calculating the cache; thread A starts to count, thread B enables a parameter tree and starts to count, thread B finishes and stores, thread A finishes and stores - cached count is now out-of-date.  This may not be the actual problem.

master has all sorts of fun locking - @meee1 says he can't reproduce there.

**Version**
ArduPlane-beta

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Bench

**Hardware type**
CubeBlack

**Logs**
n/a

tridge, 2020-12-06:
have you tried to reproduce this on master?

peterbarker, 2020-12-07:
@tridge I relied on @meee1 's reports that it didn't happen on master.

I've just tested master now and I can't reproduce using the above instructions (probably due to the extra locking mentioned above.

Realistically this isn't going to be addressed in 4.0 - so I'll remove it from Backports and close this issue.

Bring on 4.1!

closed 2020-12-07 by peterbarker