ArduPilot/ardupilot #14781: Add Support for STM32H750VB With Firmware on External Flash

davidbitton, 2020-07-08:
## Feature request

**Is your feature request related to a problem? Please describe.**
The SP RACING H7 EXTREME, ZERO, and NANO all use the STM32H750VB chip. This would be a great "30.5mm" for Ardupilot due to the availability of at least 8 motors outputs, a stackable VTX, and several other stuff.

**Describe the solution you'd like**
The support for this board is in Betaflight/Cleanflight now. The MCU shares a fair amount of functionality at the H743/H753, except internal flash is only 128k. The H750 has the ability to map external flash to internal storage banks, so you can run code from it. This all resides on a QuadSPI bus. This line of boards -- and the H750 -- would be great to have access to.

**Describe alternatives you've considered**
The alternate solution would be to use a different flight controller.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
[Seriously Pro Racing H7 Extreme](http://seriouslypro.com/spracingh7extreme)
[SPRACINGH7EXTREME Clenflight Target](https://tro.pe/3gFJwZy)

ConnorChristie, 2020-07-13:
Just got the H7 Extreme board myself and would love to see something like added to ArduPilot

haplm, 2020-11-08:
The board is super powerful, would be interesting to have this target in AP (especially for Lua scripts)

andyp1per, 2020-11-08:
QSPI is not implemented in AP. I looked at it a little and you essentially need to write another SPI driver that uses the ChibiOS qspi APIs. After that there is a bunch of work trying to organize the chip to use that.
So in summary - it's a lot of work and I doubt it will happen any time soon.
I have some interested in working on the QSPI driver if I had a board with QSPI flash - unfortunately on the Kakute F7 Mini the extra IO pins are not connected to the flash. Developing this on a board that cannot run AP in the first place sounds like unreasonable pain and suffering.

hydra, 2021-02-12:
@andyp1per The QSPI support is actually pretty easy, I think it took me less than a day to get it working, and then a bit longer to polish and clean the code, but wasn't a big issue at all.

Also, there now exists the SPRacing SSBL which allows the use of the memory-mapped mode of the W25Q flash chips as found on the SPRacingH7EXTREME PX4 edition.

https://github.com/spracing/ssbl
https://shop.seriouslypro.com/sp-racing-h7-extreme

The tricky part is being able to save the config to external flash, as the code to save to the config flash has to run from RAM, that's easy to implement by using `attribute` on the functions and then use the linker script to ensure those functions are located in RAM, in combination with some code that copies a linker section to RAM, of course.

You can see the PX4 edition PR here, to get an idea of the areas changed in PX4 to provide support for the board.

https://github.com/PX4/Firmware/pull/15385

If a developer needs assistance in getting this to work I'm available.

andyp1per, 2021-02-12:
@hydra thanks - my comment here is a little out of date. @tridge and I have a plan. As it turns out ChibiOS has good support for QSPI which we will leverage and it allows us to memory map the flash. We will also create our own bootloader to set this up for the load of the main firmware.

hydra, 2021-02-12:
@andyp1per great, the SSBL should suffice as a proof-of-concept to allow for your own second stage bootloader to work which can implement, for example, your own firmware update system or USB-MSC, or other bootloader code.

Just be aware that customers should never erase the firmware that in the CPU's internal flash on any of the SPRacingH7 boards otherwise they won't be able to boot any other firmware, to restore it customers have to send their boards back to us in the post.

Let me know if you need any assistance.

hydra, 2021-02-12:
@andyp1per Can I also draw your attention to the H7 QSPI clock mux.  Make sure that you set it, and the QSPI peripherial prescaler, correctly for the flash chip and that you set the QSPI GPIO pin frequency to their highest frequency setting.

The W25Q*JV should use up-to 133Mhz CLOCK.

https://www.winbond.com/resource-files/w25q128jv%20revf%2003272018%20plus.pdf

Use the highest clock speed for the detected flash chip for maximum code-execution performance.

Current PX4 edition boards are shipping with the W25Q128JVEIM.

andyp1per, 2021-02-12:
@hydra thanks - so what does DFU mode do then? Does it not allow you to write directly to the CPU flash?

davidbitton, 2021-02-12:
@hydra is it not possible for the Ardupilot bootloader to be used in lieu of your proprietary bootloader? I understand through our conversations that it's genesis came from using the board w/ CleanFlight. Is the ArduPilot bootloader at all any more sophisticated than the CleanFlight bl?

andyp1per, 2021-02-12:
The reason I would like to use AP's is because its totally configurable thanks to the magic of ChibiOS and our abstraction layer and I really want to put this on the MCU flash rather than in the flash chip. I would be surprised if there is a physical reason this can't be done.

davidbitton, 2021-02-13:
@andyp1per another benefit of getting the AP BL on to the H750, is that @hydra's H7 is not the only H750 on the market. @hydra would have to confirm the why, but the different winbond chip on the "PX4" version is to accommodate the manner in which the proprietary bootloader is designed.

hydra, 2021-02-13:
@davidbitton ST's MCU QSPI peripheral don't support the W25N NAND flash as they don't have the right flash command set for memory mapped mode so the PX4 edition has to use a different chip and that is the only hardware change.

The SPRacing bootloader then had to be updated to support the new flash.

The SSBL was written to bridge the gap between EXST formatted firmware and firmware that runs in memory mapped mode such as PX4/Ardupilot/etc.

The SPRacing bootloader has various features, such as SPRacing DFU mode for firmware updates, booting from SD card, updating firmware from SD card, erasing various flash sections and for new, unreleased FCs, supports loading of the PixelOSD BIOS and APIs and of course integrity checking and loading of EXST formatted firmware from the external flash.

The bootloaders supplied by manufacturers should not be erased, other FC firmware should work with whatever the manufacturer uses, not replace it.  The Radix H7 board also has a custom bootloader which supports MSC for installing firmware. Suggest talking to @mluessi with regards to whether that FC and it's bootloader supports non-betaflight firmware.

hydra, 2021-02-14:
> @hydra thanks - so what does DFU mode do then? Does it not allow you to write directly to the CPU flash?

Correct, the SP Racing bootloader DFU mode exposes the external flash, at the standard QSPI memory mapped peripheral address space which starts at 0x90000000.  The mapping of DFU mode addresses to actual memory is arbitrary, but I used that address for some consistency.  The bootloader will treat some sections of the flash as 'FIRMWARE' and some as 'CONFIG' to provide the 'erase firmware' and 'erase config' bootloader menu options, in addition to the 'erase entire flash' menu option.

Refer to the PX4 PR for firmware and config flash address offsets and lengths, I think I added details of the memory map there, if not let me know and I'll dig it out and post it here for you.  

You need three 'partitions':
* Ardupilot/PX4 code that runs directly from the flash chip.
* SPRacing SSBL or your own SSBL bootloader that provides additional functionality, aka 'Firmware partition'
* Config for Ardupilot/PX4, aka 'Config partition'

The entire 128k single flash page on the CPU is used for the bootloader.

Some additional details and menu options for the SP Racing bootloader can be found here:

https://github.com/spracing/betaflight/issues/1

andyp1per, 2021-02-15:
@hydra thanks - sounds like for now this is going to be the path of least resistance so I am going to go with it. So some questions:

I tried the obvious thing of building a version of AP with the flash offset at 0x90000000 (I also tried 0x90001000) and loading that with the SSBL. It did not work, but it seems like for it to work I have to have a specific format? Or does this only apply to the SSBL and not the firmware itself?
I was hoping that initially I could use the SSBL to get the MCU into QSPI mode, but maybe that's not going to work and I have to enable QSPI in the firmware from the get go?
Are there any indications about what has gone wrong from the SSBL? I get a solid red light.
Can you point me at the debugger pinout for the board, presumably I can attach an stlink in the normal way to debug?

andyp1per, 2021-02-21:
@hydra I'm not getting very far with this. I need the minimum possible config to at least get something working - I don't want configuration partitions, GPT or anything like that - I just want the first bootloader to boot my code at an address that is known to me. I can make a cut down version of AP fit into 448k, so for now I would like to boot into RAM like the SSBL does - then I can verify that ChibiOS is setting up QSPI correctly etc. I have looked through the SSBL code, but its not telling me a great deal since all of this setup is handled by ChibiOS for us. I've also looked through the PX4 code - again not very helpful as most of it is NuttX code that is replaced by ChibiOS in AP. Is there some kind of spec for what the spracing bootloader does or expects? For instance what memory address does the spracing bootloader bootstrap to?

hydra, 2021-02-21:
Heya,

The SSBL and Betaflight linker scripts gives you the locations of the code that the first bootloader runs, build the code and inspect the generated .map file for the actual addresses that the linker uses.

If you can build a cut-down AP in 448k then that seems like the easiest way to get something up and running.

See:

https://github.com/spracing/ssbl/blob/master/src/main/link/stm32_ram_h750_exst.ld#L58-L70
https://github.com/spracing/ssbl/blob/master/src/main/link/stm32_h750_common.ld#L16-L44

https://github.com/spracing/ssbl/blob/master/src/main/startup_stm32h750xx.s#L77

ok?

hydra, 2021-02-21:
The .map file is in build/SPRACINGH7EXTREME/RELEASE/SPRACINGH7EXTREME.map

The reset vector is defined in the vector table, here:

https://github.com/spracing/ssbl/blob/master/src/main/startup_stm32h750xx.s#L169

Example from the .map file
```
.isr_vector     0x24010000      0x298
                0x24010000                . = ALIGN (0x200)
                [!provide]                PROVIDE (isr_vector_table_base = .)
 *(.isr_vector)
 .isr_vector    0x24010000      0x298 build/SPRACINGH7EXTREME/RELEASE/src/main/startup_stm32h750xx.o
                0x24010000                g_pfnVectors
                0x24010298                . = ALIGN (0x4)

.text           0x24010298     0x2860
                0x24010298                . = ALIGN (0x4)
 *(.text)
 *(.text*)
 .text.Reset_Handler
                0x24010298       0x80 build/SPRACINGH7EXTREME/RELEASE/src/main/startup_stm32h750xx.o
                0x24010298                Reset_Handler
```

ok?

andyp1per, 2021-02-21:
Thanks @hydra I'll have another scratch on this.

My work in progress is here: https://github.com/ArduPilot/ardupilot/pull/16698 but it also requires an update to ChibiOS which is here: https://github.com/ArduPilot/ardupilot/pull/16662

davidbitton, 2021-02-24:
@andyp1per are you on Discord?

andyp1per, 2021-02-24:
Yes I am

IamPete1, 2022-09-10:
@andyp1per Looks like your PRs have been merged, can we close?

closed 2022-09-11 by andyp1per