ArduPilot/ardupilot #14789: Plane: Roll DEMAND loss of control

Naterater, 2020-07-09:
## Bug report

**Issue details**
This should never happen: in auto mode, the plane's roll demand was oscillating at 5hz at an amplitude of about 40 degrees. This will wear out servos, while causing very poor navigation performance

[image]
[image]

**Version**
Plane 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin motor electric

**Hardware type**
mRobotics X2.1-777, Drotek F9P GNSS

**Logs**
https://1drv.ms/u/s!AunBK4rE1ZDCgulvIxSXE6ADwdN-7Q?e=yGfZVQ

magicrub, 2020-07-14:
This certainly look strange. Initially I thought "reversed rudder" but that would be reactive, where this is definitely the demand going nuts.

magicrub, 2020-07-15:
I don't see anything that can cause this so I tossed the log into Mission Planner's "auto analyze" feature just to see if it sees something. Looks like it hit's some NaNs but I don't know exactly when. 
```
Test: NaNs = FAIL - Found NaN in BAT.Curr
Found NaN in BAT.CurrTot
Found NaN in BAT.EnrgTot
Found NaN in POS.RelOriginAlt
```

Maybe @tridge could take a look. Of note, it happens a couple times on the flight and I don't see anything else jump out at me that correlates.

peterbarker, 2020-07-21:
Uh.

[image]

peterbarker, 2020-07-21:
Stall prevention is on.  Vehicle's speed was all over the place.  This is enough to explain the oscillation.

So.  Why was the speed all over the place?

Airspeed sensor data looks sensible.

GPS, on the other hand:

[image]

Discontinuities in POS track:
[image]

GPS data looks fine:
[image]

POS less so:
[image]

Things going bad seem to be associated with 
```
2020-07-09 03:59:35.452 EKF3 IMU0 yaw aligned to GPS velocity
2020-07-09 03:59:35.616 EKF2 IMU0 yaw aligned to GPS velocity
```

(n.b. EKF type is 3)

Just one compass and it looks good!

[image]

... and yet rather large magnetometer innovations:

[image]

... but compass for that bit looks great.

So.  Position innovations.  Accelerometers look clean. gyros look clean
.... except for the second IMU being completely missing in the log data.

tridge, 2020-07-25:
the EKF is very unhappy, and that is what triggers it to initially fly the wrong way. Then it gets into a turn indecision, with a 180 degree turn. We need to look carefully at the turn indecision logic and work out why it didn't manage to lock into a single direction for the turn

Naterater, 2020-08-02:
To link to the "turn indecision" issue: https://github.com/ArduPilot/ardupilot/issues/13468

Maybe I'm slightly confused, but how could the EKF be so "unhappy"? Does it rely on GPS so much that it can jump hundreds of meters on a map in a split-second?

Naterater, 2020-08-19:
@tridge and @peterbarker, I have experienced this same issue half-way across the US with a slightly different setup, but it appears to be the same SERIOUS issue. It appears that we may need to change the title of this post. 

Hardware: mRo X2.1-777, plane 4.0.5, uBlox ZED-F9P. After about 1 hour of flying, the EKF gets VERY unhappy and will result in a fly-away. We recovered in FBWA and landed. We had a RTK FIX during the landing, yet the aircraft was at least 1km away on the map. 20 or so seconds after laning, the aircraft jumped on the map to the correct landing position.

I would've estimated some base station reporting issue, but the aircraft in the first event had a local base station with a fixed position, and this aircraft had a state network of base stations used for precision surveying. I have no reason to distrust the RTK injection on either aircraft when these major EKF position issues appeared. https://1drv.ms/u/s!AunBK4rE1ZDCgu1-8r6UW7NNrBiF1g?e=HwkVsQ