ArduPilot/ardupilot #14797: hwdef: KakuteF7 and KakuteF7mini motor ordering

bys1123, 2020-07-12:
This is definition in betaflight firmware:
[image]
Which makes M1 is M1, M2 is M2, M3 is M3, M4 is M4

This is ArduPilot definition:
[image]
makes the motor port hard match betaflight ordering, if user manually set FRAME_TYPE to 12 betaflightX, the ordering will be wrong. Or when user use Helicopter frame or Plane, the servo ordering is very confusing.

andyp1per, 2020-07-12:
The only problem is that by changing it anyone who is using the current motor ordering will get a shock. I'm not sure what the right thing to do is - I agree that matching betaflight makes sense, just don't know how we get there safely

IamPete1, 2020-07-12:
This is a long standing issue, i think the consern was that we would break peoples setups.

https://github.com/ArduPilot/ardupilot/pull/10363

IamPete1, 2021-08-16:
I think were too late to fix this now, we cant make a change as it would mess up everyone's existing setup. 

The current pinout is documented on the wiki:
https://ardupilot.org/copter/docs/common-holybro-kakutef7aio.html#servo-output-mapping

Its not great but I don't think we can do anything better, so I'm going to close.

closed 2021-08-16 by IamPete1