ArduPilot/ardupilot #14840: AR_WPNav: Issue with WP_SPEED_MIN unintentionally being applied in reverse

jimovonz, 2020-07-17:
## Bug report

Rover can randomly go into reverse indefinitely during an AUTO mission when WP_SPEED_MIN is set.

Also reported here: https://discuss.ardupilot.org/t/rover-randomly-backs-up-during-mission/59132/2

Pull request to fix the issue:
https://github.com/ArduPilot/ardupilot/pull/14839

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

rmackay9, 2020-12-07:
I've reproduced the issue after applying PR https://github.com/ArduPilot/ardupilot/pull/15951.  The steps are:

- create a simple mission with 1 ahead of the vehicle
- param set WP_SPEED_MIN 0.5
- arm
- acro
- rc 3 1000 (to cause the vehicle to reverse in acro mode)
- auto

The vehicle will not progress towards the waypoint but will instead keep reversing at the WP_SPEED_MIN parameter value

rmackay9, 2020-12-07:
I've got a fix for this now included in PR https://github.com/ArduPilot/ardupilot/pull/15951

rmackay9, 2020-12-08:
This is resolved in master now so closing, thanks!

closed 2020-12-08 by rmackay9