ArduPilot/ardupilot #14841: Disable GPS checks when GPS is disabled?

mcelhennyi, 2020-07-17:
https://github.com/ArduPilot/ardupilot/blob/c8f1b44ee6fb92e02df1291edbf2e35ec11acd2c/libraries/AP_Arming/AP_Arming.cpp#L1083

Here we are checking the GPS no matter what, would it make sense to add a skip around this if GPS_TYPE's are set to None, or if the EKF gps is set to not using?

rmackay9, 2020-07-17:
@mcelhennyi, yes, I think so.  We could add an accessor to the AHRS to return true/false if it is setup to use a GPS.  In the case of the EKFs that means checking the EKx_GPS_TYPE != 3.  For DCM it would probably always return true.  This could have the effect that for Plane it would always return true (because it can use the DCM) while for Copter, Rover, Sub it could return false if that param was set.

There may be other opinions out there though but I think this enhancement would make the setup easier for non-GPS environments.

muramura, 2020-07-19:
@mcelhennyi san, @rmackay9 san. Are the following changes correct?

https://github.com/ArduPilot/ardupilot/compare/master...muramura:AP_disable_GPS_checks_when_GPS_is_disabled

rmackay9, 2020-07-20:
@muramura, thanks for giving this a try.  I would do it a different way, I wouldn't touch the EKF code at all, instead I would:

1. add a new method called "may_use_GPS()" to AP_AHRS.  For DCM it would return true if "_gps_use" == true.  For the EKF2 and 3 it would return true if EKx_GPS_TYPE != 3.
2. AP_Arming's GPS check method should return true if the _ahrs::may_use_GPS() != true.

muramura, 2020-07-20:
@rmackay9 san.  Thanks for the review.
I think the fewer changes to the source, the better.
I will make changes based on comments.

IamPete1, 2021-08-16:
similar to https://github.com/ArduPilot/ardupilot/issues/14661