ArduPilot/ardupilot #14844: Different numbers being reported between the OSD and TX

DonaldHillNZ, 2020-07-17:
## Bug report

**Issue details**
The Telemetry being reported is different on the OSD and TX
_Please describe the problem_
Since the lastest Arduplane upload Ive had issues with the numbers being reported on the OSD and the TX telemetry.
The satellites being reported on the OSD seem to be correct (especially since I can arm the plane when the TX shows 0 sats) and the TX will show anywhere between 0 and 5 sats. The OSD will be showing 18 +.
Also the Alt and Air Speed (via GPS) are way out on the TX eg. alt 300ft and on the OSD 51ft. Speed around 20mph on the osd and TX 160mph.

**Version**
V 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Powered Glider
**Hardware type**
Matek 405 Wing, Crossfire Nano RX, Beilkin GPS (Standard in use on other planes not yet upgraded to V 4.0.5)

**Logs**
Nothing to show

IamPete1, 2021-08-28:
Is this still a issue? If it is can you provide some more information on the "TX" setup, Thanks.

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1