ArduPilot/ardupilot #14876: Copter: POSHOLD mode has gone beyond the polygon fence.

muramura, 2020-07-21:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
POSHOLD mode has gone beyond the polygon fence.
LOITER mode did not exceed the polygon fence.
I'm thinking, POSHOLD mode does not exceed the polygon fence, because it also uses position.

I discovered this phenomenon at SITL.

_Please describe the problem_

**Version**
_What version was the issue encountered with_

APM: ArduCopter V4.1.0-dev (717bf718)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-07-21:
@muramura, yes, that's right, the stop-at-fence feature only works in Loiter and Guided velocity mode.  It is difficult to make it work in PosHold because the pilot controls the lean angles not the velocity of the copter.  It's possible but it's a bit tricky.

chobitsfan, 2020-07-27:
I have a partial solution for this in #13720.