ArduPilot/ardupilot #14879: Copter: Internal Error 0x400 (constrained NaN) with mRo LocationOne GPS

rmackay9, 2020-07-22:
We have had two reports of Internal Error 0x400 when the mRo LocationOne GPS is used with GPS blending.

- 1st report is here: https://discuss.ardupilot.org/t/copter-4-0-0-released/50545/79
- 2nd report is here: https://discuss.ardupilot.org/t/copter-4-0-0-released/50545/95

This is also mentioned in the Copter-4.0 issues list (search for "LocationOne"): https://github.com/ArduPilot/ardupilot/issues/12547

WickedShell, 2020-07-22:
Someone probably needs to audit that everything we actually use is filled in AP_Periph. A very quick audit of it shows that `pdop` in fix2 isn't populated for example, and we do use that inside ArduPilot until we've received the aux message, which would propagate bad data in.

peterbarker, 2020-11-24:
tridge nailed this one with https://github.com/ArduPilot/ardupilot/pull/15871

closed 2020-11-24 by peterbarker