ArduPilot/ardupilot #14881: Copter: add pre-arm check that vertical velocity innovations are below 1m/s

rmackay9, 2020-07-22:
I have seen a log where the user attempted an autonomous takeoff when the EKF's velocity estimate was off by 3m/s.  The good news was the estimate was off in the safe direction (it thought it was climbing at 3m/s) so the vehicle simply didn't takeoff.

We should add a pre-arm check that the EKF thinks it's velocity estimate is correct within perhaps 0.5m/s.  In this case the user had most of the arming checks off so it wouldn't have helped them but this is another issue.

amilcarlucas, 2020-07-23:
@rmackay9 this can probably also help in [my case](https://discuss.ardupilot.org/t/copter-4-0-4-rc2-used-very-little-trust-and-went-up-to-3m-height-instead-of-the-desired-10m-height/59208) right?

rmackay9, 2020-07-23:
@amilcarlucas, yes, I think it might have helped.