ArduPilot/ardupilot #14883: QuadPlane arms with no compass (COMPASS_USEx=0)

Hwurzburg, 2020-07-22:
Plane normally allows arming with compass pre-arm checks in place if there is no compass use enabled since Plane flys nicely without a compass. However, if its a QuadPlane, this should not be allowed unless compass pre-arm check is disabled. Compass is critical to QuadPlane operation, just like Copter.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2024-07-31:
I believe we now *must* support arming without compass as we've got QuadPlanes out there that don't have one?

The Manta, I think?

Can we close this @Hwurzburg ?

Hwurzburg, 2024-07-31:
yeah, not recommended though, using Gaussian crashes QuadPlanes

closed 2024-07-31 by Hwurzburg