ArduPilot/ardupilot #14889: pixhawk Serial bug

raj241992, 2020-07-23:
## Bug report

**Issue details**
Serial (2,4,5) in Pixhawk after awhile of running give high latency. We use mavlink protocol.
I have tested Arduplane-3.9.8 and the serial worked fine.
**Version**
4.0.5
**Platform**
[*] Plane

**Hardware type**
Pixhawk v2

raj241992, 2020-07-25:
serial 2 also has high latency when connect to usb

raj241992, 2020-07-25:
This is the screen recorder for telem2 with usb .
the result was ArduPlane 3.9.11--> OK
ArduPlane 4.0.5 ---->high latency .
[Record_2020_07_25_10_.zip](https://github.com/ArduPilot/ardupilot/files/4975061/Record_2020_07_25_10_.zip)

rmackay9, 2020-07-26:
I'm not a plane person but I think one important thing to determine if the issue is caused by more messages being sent by Plane-4.0.5 vs 3.9.8.  It might be useful to look at MP's mavlink inspector to see what messages are being sent and how quickly.  This same info might be available in the log you've included though, especially if it's a tlog.

raj241992, 2020-07-28:
peterbarker said  this is an issue but we can raise SCHED_LOOPRATE parameter .  
https://discuss.ardupilot.org/t/secand-gs-serial-problem-on-arduplane-4-0-5/59400/8
when I raise the value the problem has gone .

closed 2020-10-07 by raj241992