ArduPilot/ardupilot #14904: Copter: Yaw rotation in Reuters mode does not stop in neutral.

muramura, 2020-07-25:
## Bug report

**Issue details**

I yawed and rotated in Reuters mode.
I put the yaw rotation operation in neutral.
The multicopter rotated 100 degrees from the angle of the neutral operation and stopped.
I want the aircraft to stop rotating close to the neutral angle of operation.
The ahrs.yaw_sensor value now indicates the angle.
Do I just set the value of ahrs.yaw_sensor?
I understand that the following method specifies the target angle.
attitude_control->set_yaw_target_to_current_heading()
Can I just add a method to set ahrs.yaw_sensor based on this method to deal with it?
attitude_control->set_yaw_current_heading()

**Version**
ArduCopter V4.1.0-dev (ccff0065)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD

**Hardware type**
SITL

**Logs**

[image]

rmackay9, 2020-07-26:
If the vehicle rotated too must after the pilot moved the yaw stick to neutral then I guess the issue is that the attitude controller's yaw acceleration parameter is very low or the yaw's maximum rotate rate is too high.  These parameters are:

ATC_ACCEL_Y_MAX : maximum yaw acceleration
ATC_RATE_Y_MAX : maximum yaw rotation rate

muramura, 2020-07-27:
@rmackay9 san, Thank you I'll check the parameters.

IamPete1, 2021-08-28:
If increasing `ATC_ACCEL_Y_MAX` does not help you must reduce `ACRO_YAW_P`

closed 2021-08-28 by IamPete1