ArduPilot/ardupilot #14905: EKF3 not doing 3D mag fusion

tridge, 2020-07-25:
With a copter with EK3 enabled and COMPASS_SCALE=0 the EKF is not starting 3D mag fusion. magStateResetRequest is not being set.
Possibly introduced when GSF was added?

tridge, 2020-07-25:
we never call calcQuatAndFieldStates in EKF3. That is the function that checks magFieldLearned and sets up the earth field.
It was removed when GSF went in

priseborough, 2020-07-25:
calcQuatAndFieldStates should have been removed. It's functionality is performed now when magStateResetRequest is set. We need to ensure magStateResetRequest is called when required.

priseborough, 2020-07-25:
I ended up rewriting calcQuatAndFieldStates so that ot only resets the mag field states and is called when required. See https://github.com/ArduPilot/ardupilot/pull/14906

closed 2020-07-25 by tridge