ArduPilot/ardupilot #14928: FUNCTION DIAGRAMS

logindiamant, 2020-07-28:
Dear colleagues!

**Is your feature request related to a problem? Please describe.**
Please, tell, where we could find full function diagrams of control systems?

**Describe the solution you'd like**
**Describe alternatives you've considered**
https://cache.industry.siemens.com/dl/files/911/99682911/att_863921/v1/S120_S150_List_Manual_LH1_0414_eng.pdf
Please, see p.1827-2268
It is not flight application, some technology and speed/position control

rishabsingh3003, 2020-07-28:
Hi!
The GitHub issues list is generally used for feature requests and bug reports in the codebase. 
I guess your request is more support related.
You will find more support here: https://discuss.ardupilot.org/
Feel free to open up a new topic there!

rmackay9, 2020-07-29:
We have some diagrams on our developer wiki: https://ardupilot.org/dev/docs/learning-the-ardupilot-codebase.html

For example here is the page on attitude control: https://ardupilot.org/dev/docs/apmcopter-programming-attitude-control-2.html

closed 2020-07-29 by rmackay9