ArduPilot/ardupilot #14931: Plane: flap slew rate limit does not work for flaperons and d-spoilers

IamPete1, 2020-07-28:
Flap slew-rate param only works for the dedicated flap and flap_auto servo functions. When flap is used in the flapperon and dspoiler mixer no slew rate is applied. 

Looks like a tricky fix, I think the best method would just be to refactor the slew rate servo channels function. The current function is only used by plane throttle and flaps.

magicrub, 2020-08-10:
I think @samuelctabor has fixed this on our fork which we currently working on getting into upstream soon

samuelctabor, 2020-08-17:
The problem is that the slew rate is applied in SRV_Channels rather than in servos.cpp. It's not easy to grab the slew rate limited value as the slew is done at quite a low level. In the KH fork I fixed it by replicating the slew rate limit code in servos.cpp and applying it to the argument of flaperon_update. 

A better solution would be to add a helper to get the slew rate limited output from SRV_Channels. Even better would be to distinguish between "basic" flight controls - aileron, elevator, throttle, rudder, flap and airbrake - and actuator outputs.

PR #19712 (2022-01-24): SRV_Channel: native slew limiting
This turned into a bigger rework than I had expected. ~~The key change is that the function range / angle is now a property of the function not the channel it is assigned to. That then allowed native slew rate limiting.~~ Because the limiting is done at a scaled output level rather than PWM we can now slew limit to less than 1 us per loop and inherit limiting from other functions. 

This mean flaperons can now use flap slew rate Fixes #14931

~~I hope the flash cost is not too painful, were now storing more things per function rather than per channel.~~
closed 2022-01-24 by tridge