ArduPilot/ardupilot #14936: FlowHold mode too sensitive compared to other modes

ferayQ, 2020-07-29:
Hello,
I would like to use FlowHold mode with a hereflow in my copter but the controls are too sensitive. I can't find a way to add exponential to reduce the UAV's tilt angles or a way to adjust the sensitivity of the controls. The controls are well adjusted in the other modes (Althold, loiter) but in this mode the piloting is really too sensitive while the UAV stabilizes well when all the controls are released. Why is the Flow mode more sensitive than the other modes?
I use a Herelink radio with its hereunit receiver.

rmackay9, 2020-07-29:
@ferayQ,

This seems more like a support question so it's probably best to raise it in the forums and include an onboard log if possible.

If the vehicle has an optical flow and lidar then it is best to stick with Loiter mode instead of FlowHold.  FlowHold mode's only use is on vehicle that want to do non-GPS flight using optical flow but do not have a rangefinder.

rmackay9, 2020-07-30:
By the way, I've found a small issue in the past to do with FlowHold's lean angles and recorded it here: https://github.com/ArduPilot/ardupilot/issues/10524.  I don't think it's the same issue but just for reference.

ferayQ, 2020-07-30:
> @ferayQ,
>
> This seems more like a support question so it's probably best to raise it in the forums and include an onboard log if possible.
>
> If the vehicle has an optical flow and lidar then it is [...]

Yes, I'm looking to make flights without GPS with optical flow and without range finder (the hereflow one is disabled), so it is not possible for me to use the Loiter mode. Here are the parameters of the flight card that I have configured: https://www.dropbox.com/s/5spbixzff6047hp/17.param?dl=0

lthall, 2020-07-31:
@ferayQ 

Randy politely pointed out that you are asking a support question that is more appropriately asked on the forums. The community is very helpful and active so this is the best place to ask for assistance and advice.

Issues are for known bugs or detailed feature requests. We do not answer questions or provide help in the git issue list.

I am closing this issue.

closed 2020-07-31 by lthall

rmackay9, 2020-07-31:
@ferayQ,

From the support calls I've seen, flowhold's method of determining both altitude-above-terrain and velocity just isn't reliable enough for many users.  I think the final results are going to be a lot better if a lidar is somehow used allowing Loiter and the other autonomous modes to work.