ArduPilot/ardupilot #14993: Allowed scale factor minimum too strict for RM3100?

VDLJu, 2020-08-05:
Hi,

I have seen high variation between RM3100 units on the magnetic field intensity. I have seen that the scaling factor was narrowed lately:

https://github.com/ArduPilot/ardupilot/commit/303030c524f35f790ac2f217f3619b66deacd3d3

But soon later restriction was eased for RM3100. Only the maximum scale factor limit was changed.
https://github.com/ArduPilot/ardupilot/commit/a0cf4e158abd0c4041f24e5ba7dc7f2781755a30

I'm now facing an issue where the lower limit is too strict for RM3100. 
>... bad radius 394 expected 526

This would correspond to a scale factor of 0.749, which is a lot lower than the allowed 0.85. Although this specific unit gives this error message it seems to give sensible response/measurements.

I have done this testing with AC 4.0.3 and CUAV NEO v2.
I'm not able to calibrate that specific compass due the restriction and I expect that others having calibration issues with CUAV NEO V2 could be caused by the same reason.

Similar issues
https://discuss.ardupilot.org/t/cuav-v5-nano-neo-v2-pro-gps-compass-calibration-not-possible/57140
https://discuss.ardupilot.org/t/help-with-compass-calibrations/55199

I would suggest that the compass scale factors are re-evaluated for RM3100.
I would recommend that for RM3100 the minimum should be set to 0.70.

RM3100 manufacturer (PNI) doesn't seem to give any promises over magnetometer absolute accuracy. 
However relative accuracy and linearity is great on RM3100 based on the datasheet, which I can agree on based on my own tests.

VDLJu, 2020-10-29:
Rising this issue again.
Should the minimum scale factor also to be adjusted to meet RM3100 sensor variation, discussed in #14433 and #13895.

peterbarker, 2020-11-24:
@tridge has been playing with the scale factors with f7daa4a93a3371f8e0f054c7c904928f4bc104c3 - can you check to see if that was sufficient to fix your problems, please?

IamPete1, 2021-08-28:
Seems like the issue was fixed, since there is no report the increased range is not enough.

closed 2021-08-28 by IamPete1