ArduPilot/ardupilot #15000: Head tracking input on AntennaTracker

darvin, 2020-08-06:
## Feature request

Head tracking is missing from Ardupilot ecosystem, let's add it

**Describe the solution you'd like**

Ground head tracking solution (for example, fat shark head tracker, or ark bird head tracker, or open source solution) connects to Ardupilot board with AntennaeTracker running on it. The headtracking data, together with USB joystick input connected to the Mission Planner ground station, is being transmitted to remote aircraft

**Describe alternatives you've considered**

External solution: headtracking being transmitted through crossfire module to aircraft's controller 

**Platform**
[  ] All
[ x] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2020-08-08:
I'm not really sure how Antenna tracker is involved here, I'm not familiar with the head tracking solutions but it looks like you just plug them into your transmitter, no reason why this method wouldn't work on ArduPlane.

StonedDawg, 2021-03-04:
assigning rcx_in to servo output does it, but it doesnt work very well since servo min and max limit is not very well adjustable (max at 2200 and min at 500), i can only get -60 to +60 on my 180degree servo...

closed 2021-07-26 by IamPete1