ArduPilot/ardupilot #15002: Diff All feature

InitialFPV, 2020-08-06:
## Feature request
I'm looking for an Ardupilot equivalent to "diff_all" in betaflight and inav.

This would take the full parameter list and export a list in some readable format of only the parameters that are different from their default/original values. This would be useful to diagnose problems when changing firmware or in my case when trying to diagnose a problem by simply starting from scratch. 

If this already exists I couldn't find it.

rmackay9, 2020-08-06:
It's a good idea.  I normally keep a file around which is of the defaults and then I do a diff either using one of the buttons on the right side of MP's Full Parameter Tree screen or I use GitTortoise merge.

auturgy, 2020-08-06:
In mavproxy you can use “param diff”

Regards,

James

auturgy, 2020-08-06:
https://ardupilot.github.io/MAVProxy/html/uav_configuration/parameters.html#param-diff

rmackay9, 2020-08-06:
I think it's fair to say that we don't have this request fully covered because the methods described by @auturgy and I involve the GCS which doesn't actually know what the defaults are for the vehicle and version on the autopilot.

amilcarlucas, 2020-08-12:
@auturgy That link doe not work for me. Do you mean https://ardupilot.org/mavproxy/docs/uav_configuration/index.html ?

auturgy, 2020-08-12:
No, the link was correct when I posted it. Stephen pulled docs down https://github.com/ArduPilot/MAVProxy/commit/ce8aa7b5c00b937e2297e20641462f7d7667e2e8#diff-2ed2ee3fc39e2c476fd0b5c7bfa24e69

IamPete1, 2022-08-05:
Closed by https://github.com/ArduPilot/ardupilot/pull/20972

closed 2022-08-05 by IamPete1