ArduPilot/ardupilot #15011: AP_Scripting: The luacheck tool gives me a warning of one line character over

muramura, 2020-08-08:
## Bug report

**Issue details**
I tested rangefinder_test.lua with the luacheck tool.
I found a warning for rangefinder_test.lua:28:121: line is too long (219 > 120).
I believe there is a limit on the maximum number of characters per line in the parsing of the script.
I would like to use a script checker warning, one that has no errors.
I want to know how many characters per line the LUA can handle in FC.

Checking rangefinder_test.lua                     19 warnings

    rangefinder_test.lua:6:10: setting non-standard global variable update
    rangefinder_test.lua:7:24: accessing undefined variable rangefinder
    rangefinder_test.lua:8:3: accessing undefined variable gcs
    rangefinder_test.lua:10:7: unused loop variable i
    rangefinder_test.lua:10:14: accessing undefined variable rangefinder
    rangefinder_test.lua:11:8: accessing undefined variable rangefinder
    rangefinder_test.lua:12:7: accessing undefined variable info
    rangefinder_test.lua:13:12: accessing undefined variable rangefinder
    rangefinder_test.lua:14:7: accessing undefined variable info
    rangefinder_test.lua:18:10: accessing undefined variable update
    rangefinder_test.lua:21:10: setting non-standard global variable info
    rangefinder_test.lua:22:28: accessing undefined variable rangefinder
    rangefinder_test.lua:23:24: accessing undefined variable rangefinder
    rangefinder_test.lua:24:24: accessing undefined variable rangefinder
    rangefinder_test.lua:25:18: accessing undefined variable rangefinder
    rangefinder_test.lua:26:23: accessing undefined variable rangefinder
    rangefinder_test.lua:28:121: line is too long (219 > 120) ★
    rangefinder_test.lua:28:3: accessing undefined variable gcs
    rangefinder_test.lua:31:8: accessing undefined variable update

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

WickedShell, 2020-08-08:
You don't need to open an issue when you are also about to open a PR to address it.

Line length is the smallest problem with using luacheck atm, we need a configuration file for it to make the undefined variables be okay.

closed 2020-08-08 by WickedShell

muramura, 2020-08-11:
@WickedShell san.  Comments thanks.  I understand that issue is unnecessary if I pr.  I understand that luacheck's warnings are inadequate.