ArduPilot/ardupilot #15012: ADSB on OSD

RMFPV, 2020-08-08:
## Feature request

Please add ADSB display to OSD to show nearest plane: direction with an arrow, distance and alt of that plane.

Like this:
alt25 dist100 >

**Platform**
[+] Copter
[+] Plane

**Additional context**
_Add any other context or screenshots about the feature request here._

geofrancis, 2024-04-09:
This would be really useful for avoiding other vehicles using AIS or FLARM.

drtrigon, 2026-04-01:
Even better would be to have a radar around or instead of the artificial horizont in the center of the screen and filling essentially the whole screen.

For an example, look at this: https://www.helicomicro.com/wp-content/uploads/2017/07/tbs-flarm-07.jpg