ArduPilot/ardupilot #15016: Copter Auto Takeoff Mode: Adjust The Initial Takeoff Speed

wattsinnovations, 2020-08-08:
Hello, we are finding that with larger aircraft (anything over 40lbs), the initial takeoff speed for an Auto mission is much too fast. I have posted this in the Cubepilot discuss forum and after trying everything that has been recommended, nothing is helping slow down the initial takeoff speed.

Initial Post on CubePilot Discuss forum: https://discuss.cubepilot.org/t/auto-mode-change-the-initial-spool-up-take-off-speed/1247/8

My request is to have a new parameter which allows the user to control the initial "Launch" speed. It seems that the copter first goes through an initial takeoff protocol, then as soon as it hits maybe 10' or so in altitude, it then begins to climb at a rate set by WPNAV_SPEED_UP. I am proposing that a parameter be added to slow down the initial spool up/takeoff process.

Because this takeoff process happens so fast for such a large aircraft, it will often output two errors: EMERGENCY Potential Thrust Loss and EKF IMU1 Ground Mag anomaly, yaw re-aligned.  

Anything we can do to assist with this would be great. 

Thank you!
Bobby

amilcarlucas, 2020-08-12:
Simply reduce the acceleration parameter, and the speed will slowly increase. Works great here.

wattsinnovations, 2020-08-12:
Hello, thank you for the reply. Are you referring to WPNAV_ACCEL_Z? If so, that parameter did not adjust the initial takeoff vertical speed, it only adjusted the speed at which the drone climbs after it has popped up into the air.

IamPete1, 2021-08-16:
Sounds like this was resolved by the setting of `MOT_SPOOL_TIME`

https://discuss.cubepilot.org/t/auto-mode-change-the-initial-spool-up-take-off-speed/1247/16?u=iampete

closed 2021-08-16 by IamPete1

rmackay9, 2021-08-17:
I think there was definitely a bug in 4.0.x re the aggressiveness of the takeoff that has been fixed in 4.1.  I've reviewed the Copter-4.0.x release notes and I don't think the fix was included in 4.0.