ArduPilot/ardupilot #15025: Baro: altitude error is linear to altitude

magicrub, 2020-08-11:
For high-altitude flights the baro seems to always have a linear error that is linear to altitude. The easy solution for this is to set EK2_ALT_SOURCE = 2 (GPS). However, it would be worthwhile to trace down the root cause.

Possible problems:
- temperature measurement is not correct
- ground calibration assumptions (if any?) are not correct
- formula itself (pressure to meters) needs improvement

vierfuffzig, 2020-09-26:
@magicrub i'm seeing fairly constant linear errors of baro vs GPS (using additional external baros) with:
- the onboard and external baro differing no more than 0.1 - 0.3%, regardless of relevant temp differences (20 - 40°C), regardless of sensor type used (MS5611 / BMP280)
- barometric altitude constantly 4 - 6% higher than GPS altitude
- the onboard baro always reporting the highest altitude
- usual lauch point being at ~ -3 m ASL, baro alt constantly diverges with increasing pos offset. when launching from an elevated place, baro and GPS first converge, until baro alt supersedes GPS alt and reaches the usual ~ +5% offset at altitude

this is constant across logs generated over >2 yrs on various board types, usually using master codebase not more than a couple of weeks old. looking at the nearly, but not perfectly constant error i've been assuming there's both an offset and a gain involved.

IamPete1, 2021-07-26:
https://github.com/ArduPilot/ardupilot/pull/17984