ArduPilot/ardupilot #15053: Completion Waf: complete --target

khancyr, 2020-08-14:
## Feature request

**Describe the solution you'd like**
We should completeing --target parameter on waf for easier build.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Notorious811, 2020-09-25:
i am not much familiar with using waf build process, but i can work around with those wscripts, i'm just not able to get what we really looking here by saying "completeing --target parameter on waf ", can you please describe a little

khancyr, 2020-09-26:
If you are on bash or zsh you may have install our shell completion system : the thing that suggest word when you push the tab key. The support on waf is really basic and it would be good to add completion for the different target we have on waf.

Notorious811, 2020-09-26:
Gotcha, like a custom bash tab completion script for waf build options for different program groups, i can give it a try

NinaPacifier, 2020-12-12:
@khancyr I am new to open-source contributions and want to contribute to the ArduPilot organization. Can I work on this issue???

khancyr, 2020-12-17:
@NinaPacifier sure, we don't have strict rules on assigning issue. So feel free to solve the ones you want to do !

khancyr, 2021-01-07:
closed by https://github.com/ArduPilot/ardupilot/pull/16206

closed 2021-01-07 by khancyr