ArduPilot/ardupilot #15054: Alt controller bug during RTL

buckker, 2020-08-14:
the flightcontroller does not hold the target altitude during RTL. I noticed this behavior on several drones over 200 flights. The drone climbs or descends a few meters on the last waypont and fly on this incorrect altitude to the start point. There is no sign, that the flighcontroller try to adjust the  altitude to the target altitude. This could be potentially dangerous, if the incorrect altitude leads to a crash we obstacles. 

The bug appears with 4.0.3 and 4.0.4

I allready posted the bug here: https://discuss.ardupilot.org/t/auto-mission-climb-after-last-wp-rtl/58362/5

From a flight today: 

[image]

The mission in Mission Planner looks like that:
[image]

[00000109.zip](https://github.com/ArduPilot/ardupilot/files/5074818/00000109.zip)

lthall, 2020-10-21:
This is because you have set your RTL alt to 3m (300). I suspect you meant to put it in as 30m (3000).

It should not change height but it does because the height reference frame changes but the height does not.

IamPete1, 2021-07-26:
Sounds like this was a config error.

closed 2021-07-26 by IamPete1