ArduPilot/ardupilot #15070: During AUTO mission dron stopped at 1st waypoint and dont obey the command of going to 2nd waypoint

jmglezotero, 2020-08-16:
Hi guys,

This if the first time I am reporting a bug, so I hope I am doing it correctly.

The problem arised with an Arducopter V 3.6.9 running on a Cube Blue while flying a typical AUTO mission, very similar to the plenty of missions we had carried out so far with the same dron. The dron was heading to the 1st waypoint, normally, and when arrived it stayed there and never goes to the 2nd waypoint. When you check the log, you can see that the command to go to the 2nd waypoint was actually recorded… however, as I said, the dron never moved from the 1st waypoint. It hovered there for more than 1 minute, until the pilot realized that something was wrong and commanded RTL, a command that the dron this time obeyed. We posted this issue on the forum, where you can download the log…

https://discuss.ardupilot.org/t/unexpected-behaviour-in-mission/60312

I think this is a firmware bug because there were not any message on the cockpit saying that something was wrong… the dron just did not want to obey the pre-stored command of going to the second waypoint. I have been reading on the forum, and at least two guys have called the attention to something similar to this, with previous firmware versions. Here are the posts…

https://discuss.ardupilot.org/t/auto-mission-hang-at-first-second-waypoint/18599

https://discuss.ardupilot.org/t/copter-intermittently-stops-between-wps-and-just-slowly-climbs/18778

It seems that it may be a sort of problem when you have a serie of Do commands (like conditional yaw) and Wp cpmmands… but it is weird because I have been flying with conditional yaws all my life and this is just the first time something like this happens to me…

I also reviewed in the release notes to see if this issue is something that have been corrected in future firmware releases… but cannot see any release notes that resembles this case… In case this is something that you guys, the developers, are aware off and it has been fixed, I would appreciate if you can point to me to the exact firmware version in which the issue was fixed and reported in the release notes. If by the contrary, this is something that have not been addressed, I would like that you can take a deep look to this case… If there is something in which we can help, please let me know.

Thanks in advance

GiovaneMunhoz, 2021-03-16:
@jmglezotero 

Hello friend, I had this same problem twice, but today I've found the solution.

I don't know why,, my FC has changed the param WPNAV_RADIUS to 0 in both time, this value can't be zero it has to be between 5 to 1000cm, this way the copter cant reach the waypoint and to to second WP.

Today I did the test, setted it to 100 and it flew percetly and using the same mission I changed to zero and the same problem happen, got back to 100 and flew ok again without changing mission.

rmackay9, 2021-03-16:
@GiovaneMunhoz 

Thanks for the explanation.  I suspect it in the ground station that has changed the parameter value and it seems easy to do because Mission Planner highlights the field by default when the user first switches to the Plan screen.  I've raised an issue to have MP changed: https://github.com/ArduPilot/MissionPlanner/issues/2576.

I think I will close this issue because it really is a support issue I think.

closed 2021-03-16 by rmackay9