ArduPilot/ardupilot #15083: bug in handling of loss of primary compass

tridge, 2020-08-18:
If primary compass is lost then the EKF does not fall back to a backup compass

rmackay9, 2020-08-18:
I agree that users would expect the EKF to fall back to the 2ndary compass in this situation.

As mentioned on the dev call, when we resolve this issue we will need to be careful especially with Copter users to ensure they test with the internal compasses if they plan to leave them enabled.  As we all know, internal compass can get a lot more interference and after this fix goes in, these backup compasses could be used while in previous versions they never would have been.  Perhaps some extra advice on the wiki to explain to users how to test flying with their backup compasses will help and then if they find they don't work well, they can disable them.

OXINARF, 2020-08-18:
@WickedShell has brought this issue to my attention. I just want to say this isn't new knowledge (for the _USE part we even have an issue for it, #10785). We have a PR open (#6912) regarding these issues - please read it carefully before attempting a fix for this.

Let me add that this shouldn't be just with primary, if EKF switched to second compass for some reason and then that compass was lost, I expect that it also wouldn't change.

rmackay9, 2020-08-18:
@OXINARF, great, thanks very much for the extra info and links.

fr-correia, 2020-10-07:
Hi guys, just passing by.
I've had some problems with mags in the past and I may give you something new:
If you want to kill compasses and force a switch between them, just start the flight(mags fusion may not work while on the ground), and pop some big offsets on the current mag. The innovations will grow above 1 for 10s if you insert the maximum values, (-400 or 400) in any of the components.

Also I found something interesting, whenever there is a switch of compasses, the boolean variable magStateReset is put to true, and can only be put to false again in the function controlMagYawReset(). In the cases of the airplanes the function returns always in the first condition condition of that functions and never really resets the magStates, which disables the fusion of mag data.

Hope i could be of any help! ;)