ArduPilot/ardupilot #15090: AP_HAL_ChibiOS: F35Lightning / 10DOF WingFC support broken

vierfuffzig, 2020-08-18:
Testing current code against plane 3.9 in response to a support request on discuss revealed that board support seems currently broken, possibly some issue with MPU9250 defines. 
symptoms: board deliberately getting stuck in init, random acc / gyr motion, acc cal fails.
works as expected with plane 3.9.
will do bisect as soon as time permits.

**Version**
tested on plane 4.0.5 + master

**Hardware type**
F35Lightning / 10DOF WingFC

**Logs**
no log hardware on that board type

vierfuffzig, 2020-08-18:
a bisect search identifies https://github.com/ArduPilot/ardupilot/commit/9c2caf5b121f26a19e3ee5a7d0016e54d8b17fc1 (IMU fast sampling) as the commit causing issues. it's beyond my scope to tell what the exact issue with fast sampling on the MPU9250 is though.

peterbarker, 2020-08-18:
On Tue, 18 Aug 2020, vierfuffzig wrote:

> a bisect search identifies 9c2caf5 (IMU fast sampling) as the commit causing issues. it's beyond my scope to tell what the exact issue with fast sampling
> on the MPU9250 is though.

That's a very old commit.... were you realy running such an old firmware? 
I'm sure I've put master on my WingFC here and it hasn't shown problems 
*after* that commit went in.

We've had some stuf go in to do with fast sampling recently... I'm 
wondering if the bisect might have pointed to the wrong commit.

In particular, d4ba8212970b36180c9173fe538223cf39991761 deals with fast 
sampling.

peterbarker, 2020-08-18:
@vierfuffzig well, that was really easy to reproduce; my board had fast sampling off - on made things all kinds of nasty.

vierfuffzig, 2020-08-19:
@peterbarker 

> That's a very old commit.... were you realy running such an old firmware?

it was in response to another user's support request on discuss.

> I'm sure I've put master on my WingFC here and it hasn't shown problems *after* that commit went in.

yes, same for me. it needs current param defaults (FAST_SAMPLE = 1) for this issue to take effect, which possibly is not the case if you successively update from a pre-nov2019 firmware keeping your parameters.

> In particular, [d4ba821](https://github.com/ArduPilot/ardupilot/commit/d4ba8212970b36180c9173fe538223cf39991761) deals with fast sampling.

it's https://github.com/ArduPilot/ardupilot/commit/9c2caf5b121f26a19e3ee5a7d0016e54d8b17fc1 that actually makes it default to "on" via compile-time define

> well, that was really easy to reproduce

yes, retrospectively. to me it wasn't that obvious while i had no idea what i was looking for though.
it needs some more testing, but as for now i don't think we can definitely rule out that fast sampling requests an unsupported rate on all MPU9250s. @andyp1per reported not seeing any issues on his pixracer's MPU9250 though.

Moksh-2000, 2020-08-30:
> Testing current code against plane 3.9 in response to a support request on discuss revealed that board support seems currently broken, possibly some issue with MPU9250 defines.
> symptoms: board deliberately getting stuck in init, random acc / gyr motion, acc cal fails.
> works as expected with plane 3.9.
> will do bisect as soon as time permits.
> 
> **Version**
> plane 4.0.2 and later + master
> 
> **Hardware type**
> 10DOF WingFC
> 
> **Logs**
> no log hardware on that board type

I'm sorry for wrong side here , but I needed help , recently I design new custom frame design but how to compile to .cpp to apj file and how to upload to pixhawk please help me.
Thank you 🙂

vierfuffzig, 2020-08-30:
@Moksh-2000 see dev wiki for how to set up a build environment https://ardupilot.org/dev/docs/building-the-code.html
and please use https://discuss.ardupilot.org/ for support requests.

vierfuffzig, 2020-09-06:
fixed by https://github.com/ArduPilot/ardupilot/pull/15202
closing

closed 2020-09-06 by vierfuffzig