ArduPilot/ardupilot #15098: WIP: Missing ArduPilot mavlink support to light up QGC features

DonLakeFlyer, 2020-08-18:
I'm still working on building the list...

### Click to Orbit support
* `MAV_CMD_DO_ORBIT`, `MAVLINK_MSG_ID_ORBIT_EXECUTION_STATUS`
[image]

### New ROI commands
* `MAV_CMD_DO_SET_ROI_LOCATION`, `MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET`, `MAV_CMD_DO_SET_ROI_NONE` replace the single `MAV_CMD_NAV_ROI` command.
* This is mostly a reorganization of a single one size fits all command into more specific individual commands.
* This lights up these features in QGC:
  * Click in map to set ROI from fly view as well as cancel
  * Simpler creation of missions which use ROI. You don't need to show the user some crazy complex pile of settings to support old `MAV_CMD_NAV_ROI`
[image]
[image]
  * Structure Scan complex pattern in Plan. This needs the new `MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET` support to work.
[image]

### Camera and Gimbal commands for simpler mission planning
* `MAV_CMD_SET_CAMERA_MODE`, `MAV_CMD_DO_MOUNT_CONTROL`, `MAV_CMD_IMAGE_START_CAPTURE`, `MAV_CMD_IMAGE_STOP_CAPTURE`, `MAV_CMD_VIDEO_START_CAPTURE`, `MAV_CMD_VIDEO_STOP_CAPTURE`, `MAV_CMD_DO_SET_CAM_TRIGG_DIST` 
* These allow for a simpler ui for manipulating a camera/gimbal. In QGC this capability is available on every waypoint you add. Here is what it looks like with the camera section expanded:
[image]

### Follow Me
* `FOLLOW_TARGET`
* This allows a gcs to implement follow me in a generic way

### Ready to Fly
* `MAV_SYS_STATUS_PREARM_CHECK`
* Shown here: https://github.com/mavlink/qgroundcontrol/pull/8968

### ArduPilot dialect messages:
* `MAVLINK_MSG_ID_CAMERA_FEEDBACK`, `MAVLINK_MSG_ID_WIND`
* QGC currently supports these but it would be nice if ArduPilot also spit out the common versions of these messages so QGC doesn't need to have any specific ArduPilot dialect stuff in it.

### Surveys which capture images right up to the survey edge
* `MAV_CMD_CONDITION_GATE`
* If you just use a waypoint at the end of a transect you can't really trust it to the stop camera triggering at a reasonable place. This is due to the mission state machine moving to the next command once it hits acceptance.
* Because of those most survey need to be planned with additional turnaround even for multi-rotor that go past the survey edge. And also you have to capture images in these turnaround as well to get good coverage over the actual survey area you want.
* This in turns generates a lot of crap images outside your survey area you don't really want.
* The condition gate command holds the mission state machine such that it want advance until the vehicle fly past an orthogonal line. This allow you to the stop camera triggering after that. Which in turn allows you to only trigger the camera in side the portion of the transect which is inside the survey area. Just generating way less crap data to process.

### New Mavlink Camera microservice
* https://mavlink.io/en/services/camera.html
* This allows a gcs to not only know if the vehicle has cameras available but also what feature the camera(s) supports.
* Which in turn allows a gcs to procide a rich ui to the user: https://github.com/mavlink/qgroundcontrol/pull/9003

DonLakeFlyer, 2020-08-18:
Added `COMPONENT_INFORMATION`

khancyr, 2020-08-19:
hello,

Thanks for the list ! Some point are really interesting.

DonLakeFlyer, 2020-08-19:
I pulled off the WIP. I think I have most everything listed now.

Sextant, 2020-08-19:
Thanks for posting this DonLakeFlyer, I am very interested in the Click to Orbit support.

olliw42, 2020-08-20:
@DonLakeFlyer 
I shouldn't ask here since unrelated to AP but to QGC, yet it's triggered by this here.
As regards
> Camera and Gimbal commands for simpler mission planning
> MAV_CMD_SET_CAMERA_MODE, MAV_CMD_DO_MOUNT_CONTROL, MAV_CMD_IMAGE_START_CAPTURE, MAV_CMD_IMAGE_STOP_CAPTURE, MAV_CMD_VIDEO_START_CAPTURE, MAV_CMD_VIDEO_STOP_CAPTURE, MAV_CMD_DO_SET_CAM_TRIGG_DIST

are these send by QGC as broadcast or with target? (i.e. are they reached by flight controller, gimbal, and camera)(is relevant to know to prevent multiple recipients acting simultaneously)

DonLakeFlyer, 2020-08-20:
This is for missions. A mission is targeted to the autopilot component of the vehicle as a whole. This is not COMMAND_LONG. What the vehicle does with them eventually as far as routing is up to the firmware.

The other usage is for the mavlink camera microservice but that is a different thing.

olliw42, 2020-08-21:
> This is for missions. A mission is targeted to the autopilot component of the vehicle as a whole. This is not COMMAND_LONG. What the vehicle does with them eventually as far as routing is up to the firmware.

ah. Thx. I'm kind of always thinking that a CMD should be supported both in mission and as message, but I see now that you were talking about the first while I was concerned about the second. Thx for clarifying.

DonLakeFlyer, 2020-08-25:
Added `MAV_CMD_DO_CHANGE_SPEED` problesm:
* https://discuss.ardupilot.org/t/auto-mission-hover-speed/60849
* https://discuss.ardupilot.org/t/cant-change-default-speed-in-survey-or-wp-mission/60000

amilcarlucas, 2020-12-15:
ArduCopter 4.0.6-rc1 version will have "Ready to Fly" using MAV_SYS_STATUS_PREARM_CHECK support :)

DonLakeFlyer, 2020-12-17:
> ArduCopter 4.0.6-rc1 version will have "Ready to Fly" using MAV_SYS_STATUS_PREARM_CHECK support :)

Awesome, thanks. I had some hacks in there for ArduPilot to make it work at least poorly without this. I'll remove them.

kovacsadam1986, 2021-02-27:
HelloHello everyone!
Sooo helpfully this forum, and this new mobile qgc app /4.1.1- downloaded.

I didnt know where to post it- but saw DonLakeFlyer is goodboy and  helpful- i have one question!

I use boat in rover mode- all settings ok but i can't save my .param file and can't save my. plan file anyone have idea why???
//* Plan save error--- no such file or directory and other message in another phone is a permissions denied?!?! 

My old-old phone work proper but just?!!?

I remain respectful-- and many thanks if anyone can help me.. :/

hendjoshsr71, 2021-06-19:
BATTERY_STATUS.charge_state reporting has been merged

amilcarlucas, 2021-09-21:
MAV_CMD_DO_CHANGE_SPEED should also work in 4.1.0

drew-vision, 2025-06-25:
+1 for these features. I see structure scan is still not supported. MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET is reporting an error

peterbarker, 2026-03-22:
> +1 for these features. I see structure scan is still not supported. MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET is reporting an error

This is just waiting for someone to test it: https://github.com/ArduPilot/ardupilot/pull/23092