ArduPilot/ardupilot #15176: After calibrate pressure I get "Prearm: internal error (0x8000)"

gezzuzz, 2020-08-27:
**Issue details**

After calibrate pressure I get "Prearm: internal error (0x8000)"

**Version**
4.0.0 and up

**Platform**
[X] Submarine

**Hardware type**
Pixhawk 1 and Pixhawk 4 mini

**Logs**
After calibrate pressure I get "Prearm: internal error (0x8000)"
I would have to reboot to get error to clear.
I was able to fix the issue by commenting out lines 215-225 of AP_Baro.cpp
error 0x8000 is main loop stuck.. 
Seems this code is locking up the main loop

`    for (uint8_t i = 0; i < 10; i++) {
        uint32_t tstart = AP_HAL::millis();
        do {
            update();
            if (AP_HAL::millis() - tstart > 500) {
                AP_BoardConfig::sensor_config_error("Baro: unable to calibrate");
            }
            hal.scheduler->delay(10);
        } while (!healthy());
        hal.scheduler->delay(100);
    }`

Had issue with bar30 sensor keller sensors. 
several version of ardusub 4.0.0
Using both pixhawk1 and 4 mini

peterbarker, 2020-08-28:
On Thu, 27 Aug 2020, gezzuzz wrote:

> After calibrate pressure I get "Prearm: internal error (0x8000)"

Does this reproduce on ArduPilot master?

jaxxzer, 2020-09-14:
Confirmed this problem on ArduSub 4.0.1
The issue does not happen on master (0bb44ba9)

This should be patched for ArduSub 4.0.2

peterbarker, 2020-09-21:
Backport fix was merged for this - thanks @gezzuzz

PR #15340 (2020-09-21): GCS_MAVLink: fixed critical error on prefight cal ops
fix #15176
closed 2020-09-21 by peterbarker