ArduPilot/ardupilot #15179: Custom firmware build server

tridge, 2020-08-28:
We are quickly getting to the stage that we can't fit new features on boards with 1M flash. These boards are very common, so this is becoming a critical issue.
I think the solution is to create a custom build server for ArduPilot. The custom build server would allow users to select the features they want to enable for their board and would allow them to build a custom firmware with the selected features.
The build server would be somewhat like https://terrain.ardupilot.org/, but for firmwares. Key inputs from the user would be:
 - what vehicle type (copter, plane, rover, heli, sub, tracker)
 - what board type
 - what release type (stable, beta, unstable/master)

Once those are selected the user would get a list of features. Each feature would have a tick-box next to it, and an estimate of the flash cost of that feature in kilobytes. A estimated total flash size would be displayed, along with the amount of flash available on the selected board (taking into account the storage sectors etc).
At the bottom would be 3 buttons:
[Defaults]: reset the selected options to defaults for this board
[Download]: download the build, if there is a cached build for the selected options
[StartBuild]: add the build to the queue of pending builds
The build queue would be processed by a separate daemon, and would create a build with a build like Copter-KakuteF7-GITHASH-optionXXXXXXXX.apj which would include a hash of the selected options in XXXXXXXX.
There is quite a lot of work needed for this service. Key things needed are:
 - [ ] a format for selected build options, probably something like a text file with HAL_WITH_XXFEATURE=1/0 on each line
 - [ ] a configure option to supply this build options file (could be tacked onto the end of the hwdef.dat before processing?)
 - [ ] a file containing the list of selectable features and their approx flash cost (maybe JSON format?)
 - [ ] a script to generate the list of selectable features with flash cost
 - [ ] the build server web UI
 - [ ] the build server build processor

This issue is for gathering ideas on how to tackle this task and discussing progress

rmackay9, 2020-08-28:
At the risk of adding to the work, it might be useful if users can:

1. see the build errors in case the build fails
2. upload or copy-paste in a list of parameters and default values
3. add a lua script to be included in the ROMFS
4. customise the firmware name?

peterbarker, 2020-08-28:
On Thu, 27 Aug 2020, Andrew Tridgell wrote:

>  *  a format for selected build options, probably something like a text file with HAL_WITH_XXFEATURE=1/0 on each line

Major issue I see is dependencies.

The sanity checks in the botton of Copter's config.h only partly cover 
them.

stephendade, 2020-08-29:
I have a few thoughts around support and testing.

-  Are these generated firmwares going to be run through the autotest (or any other testing?), or will it be "at the user's own risk"?
-  To assist with user support, we would a list of enabled/disabled modules to be logged (via statustext?) on startup
- Are these builds based on current stable? Or is the user able to select master or prior stable versions?

My major concern is the testing/support overhead this may produce. We need to figure out what sort of build/autotest and real life testing these custom firmwares get, as some combinations of disabled/enabled modules can cause dependency issues. Example of this is the ICEngine module depending on the RPM module.

I suspect a first step would be to see how much can actually be stripped out of a 1Mb firmware and still have something flyable/drivable.

As an alternative, perhap we could have an "FPV" build of ArduPilot that includes features that a typical FPV pilot would want and excludes features they'd typically not want.

andyp1per, 2020-08-30:
To build on what Stephen was saying, I think the idea of profiles would be highly useful. I really doubt there is a huge number of variants - and in an ideal world we would record and test the main ones. 

Also many features are currently enabled by module, whereas what you actually want is specific drivers enabled.

stephendade, 2020-08-31:
Building on my previous comment, I think it'd be best from a testing/safety perspective that we have a few profiles:

- FPV
- Outdoor Flight
- Indoor Flight

I think this would cover most use cases and mean the CI/Autotest can keep track of the builds.

With the custom build server:
My primary concern is that, once the dependencies between modules are initially fixed, there's nothing to automatically check that they remain fixed. If a new commit subsequently breaks this, the first people to discover this would be the users flying it on their vehicles.

WickedShell, 2020-08-31:
> a script to generate the list of selectable features with flash cost

Don't you have to build all the various options for this, and isn't it not actually constant, because options quickly start to interact?

khancyr, 2020-09-03:
I guess something like that would help : 
on SITL : 
```
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  78.2%  26.0Mi  76.6%  2.77Mi    [599 Others]
   5.1%  1.69Mi   0.1%  3.90Ki    ../../Rover/AP_Arming.cpp
   2.7%   933Ki   4.2%   154Ki    ../../libraries/AP_UAVCAN/AP_UAVCAN.cpp
   1.2%   418Ki   1.5%  54.4Ki    ../../libraries/AP_GPS/AP_GPS_UAVCAN.cpp
   1.2%   404Ki   2.2%  83.1Ki    ../../libraries/GCS_MAVLink/GCS_Common.cpp
   1.1%   371Ki   0.0%       0    [section .debug_aranges]
   1.0%   344Ki   1.3%  49.7Ki    ../../libraries/AP_UAVCAN/AP_UAVCAN_DNA_Server.cpp
   1.0%   335Ki   1.4%  50.8Ki    ../../libraries/AP_HAL_SITL/CANSocketIface.cpp
   1.0%   329Ki   2.1%  77.8Ki    ../../Rover/Rover.cpp
   0.9%   320Ki   0.8%  28.8Ki    ../../libraries/AP_CANManager/AP_CANTester.cpp
   0.8%   265Ki   0.4%  15.4Ki    ../../libraries/AP_CANManager/AP_CANManager.cpp
   0.6%   216Ki   2.3%  85.0Ki    ../../libraries/AP_NavEKF3/AP_NavEKF3_core.cpp
   0.6%   216Ki   2.1%  78.7Ki    libraries/AP_Scripting/lua_generated_bindings.cpp
   0.6%   210Ki   0.5%  18.8Ki    ../../libraries/AP_Arming/AP_Arming.cpp
   0.6%   207Ki   0.5%  18.3Ki    ../../libraries/AP_Airspeed/AP_Airspeed.cpp
   0.6%   206Ki   1.7%  64.6Ki    ../../libraries/AP_HAL_SITL/HAL_SITL_Class.cpp
   0.6%   204Ki   0.4%  16.4Ki    ../../libraries/AP_Compass/AP_Compass_UAVCAN.cpp
   0.5%   182Ki   0.5%  19.9Ki    ../../Rover/GCS_Mavlink.cpp
   0.5%   179Ki   0.4%  15.7Ki    ../../libraries/AP_Baro/AP_Baro_UAVCAN.cpp
   0.5%   164Ki   0.3%  11.2Ki    ../../libraries/AP_RangeFinder/AP_RangeFinder_UAVCAN.cpp
   0.5%   162Ki   0.6%  20.4Ki    ../../libraries/AP_Compass/AP_Compass.cpp
 100.0%  33.2Mi 100.0%  3.61Mi    TOTAL

```

IamPete1, 2021-08-08:
https://github.com/ArduPilot/ardupilot/issues/773

IamPete1, 2021-09-12:
Server is live, can we close? Or at least update some of the tasks..

https://custom.ardupilot.org/

khancyr, 2021-09-14:
yep I think we could close this and let the custom build server todo list be on its own project

closed 2021-09-14 by khancyr