ArduPilot/ardupilot #15181: Does the code support cars with Mecanum wheel?

liu-jiaming, 2020-08-28:
## Feature request

I'm working on the Mecanum wheel，I hope APM can support it.

**Platform**
[  ] Rover

**Additional context**
Here's a description of Mecanum Wheels：
https://www.vexrobotics.com/mecanum-wheels.html

jmachuca77, 2020-08-28:
https://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html#omni-vehicles

liu-jiaming, 2020-08-28:
> https://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html#omni-vehicles

This is my car,it's the second type of page, right?

[image]

[image]

closed 2020-08-28 by IamPete1

rmackay9, 2020-08-29:
@liu-jiaming, yes, it should be the middle one.  The OmniX.

By the way, AP's support of Omni vehicles is not quite complete.  So you'll find you can move the vehicle laterally in manual mode but not in other modes.  So for example in Autonomous modes the vehicle will continue to steering back towards the line between waypoints instead of moving sideways towards the line.

jordancormack, 2024-10-19:
@rmackay9 is that still the case?

I'd like to build something with a similar motor configuration, but want to move laterally in auto mode, without changing heading.

geofrancis, 2024-10-19:
yes unfortunately.