ArduPilot/ardupilot #15200: Plane: TECS Integrator winds up during Pitch (Nudge) demands in Auto modes (RTL, etc)

JoshWelsh, 2020-09-01:
## Bug report

**Issue details**

In discussion with Tridge, we noted an issue in Plane where if a vehicle is in RTL or Auto and the Operator issues a Pitch nudge from the RC, (for example, pitch down to reduce altitude) the throttle will spin up due to TECS trying to get the Plane back on Track.  When the operator let's up on the nudge, the plane will jerk suddenly and return back to the previous track - which in this example is higher than the operator wants the current RTL to happen.

**Version**
All versions of Plane

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane of any kind

**Hardware type**
All

**Logs**
Tridge has the logs

JoshWelsh, 2020-09-01:
We discussed a fix that would set a flag so the integrator knew that the operator was issuing a pitch demand.

tridge, 2020-09-01:
yep, we need to tell TECS not to wind up the integrator when pitch stick applied

JoshWelsh, 2020-09-01:
> yep, we need to tell TECS not to wind up the integrator when pitch stick applied

Will that also reset the track so that the planned descent happens from the new altitude and not the previous altitude?  Apologies, I forget if I asked that on the call.