ArduPilot/ardupilot #15204: Cube Orange / Copter 4.0.3 : STAT_BOOTCNT , STAT_RUNTIME and STAT_FLTTIME stays at 0

GlidLov, 2020-09-01:
Platform:  Copter  4.0.3

board: Cube ORANGE

just received my first Cube Orange, flashed official copter 4.0.3 and noticed that   , STAT_RUNTIME and STAT_FLTTIME stays at 0 and STAT_BOOTCNT stays at 1.
Previous setup with the cube black was counting correctly.

peterbarker, 2020-09-02:
@GlidLov your Cube needs to see a time source of some description before it will start to persist those parameters.  This can be either a GPS or a message from the GCS specifying the current time of day.

Please reopen this issue if supplying a time to the autopilot stops it persisting those parameters.

closed 2020-09-02 by peterbarker

GlidLov, 2020-09-02:
Fixed, once a GPS receiver is connected, STATS begin to run correctly. Issue can be closed, thanks

rmackay9, 2020-09-02:
I find it a bit odd that we have this restriction on requiring a GPS though... the boot count surely shouldn't require a GPS .. and the times are all relative..