ArduPilot/ardupilot #15217: STAT_FLTTIME does not start with 0 seconds after arming

rrr6399, 2020-09-02:
## Bug report

STAT_FLTTIME does not start at 0 seconds after arming, but rather with numbers like 783360.  

**Issue details**

I would like to know how long the drone has been flying even if the GCS has to be restarted. The only parameter that I found that tracks that is STAT_FLTTIME, but it doesn't get initialized to 0 at the beginning of the flight so I have to take a delta between the current STAT_FLTTIME and the one received after arming. If the GCS is restarted, I lose the reference point. (I could persist the reference time to a file, but I shouldn't have to do that.)

It appears that deltas are taken in relation to _flying_ms in the AP_Stats.cpp:80 file, but there is an error someplace in the bookkeeping.

(If there is some other MAVLink message that has the correct flight time, please document that here)

**Version**
_What version was the issue encountered with_
ArduCopter 4.0.4

**Platform**
[ ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Orange Cube

**Logs**

WickedShell, 2020-12-08:
`STAT_FLTTIME` is intended to be your cumulative flight time of the autopilot. It's not intended to reset between boots, or on arming. That would have to be a separate parameter or similar. There could be a new message potentially to send you that time, but it would actually be an abuse of the protocol to stick something that ephemeral into a parameter (as well as wear on storage)

peterbarker, 2020-12-08:
On Tue, 8 Dec 2020, WickedShell wrote:

> STAT_FLTTIME is intended to be your cumulative flight time of the autopilot. It's not intended to reset between boots, or on arming. That would have to be
> a separate parameter or similar. There [...]

We do have this data around for some vehicles.  Perhaps we could add an 
extension to an existing mavlink message to transport it?

WickedShell, 2020-12-08:
Am extension would be nice as there are things I could use this for in the
GCS. I do a timer at the moment but it's based off the MAV_STATE and gets
messy once a failsafe is hit, or if the GCS is restarted...

peterbarker, 2020-12-09:
On Tue, 8 Dec 2020, WickedShell wrote:

> Am extension would be nice as there are things I could use this for in the
> GCS. I do a timer at the moment but it's based off the MAV_STATE and gets
> messy once a failsafe is hit, or if the GCS is restarted...

VFR_HUD?

WickedShell, 2020-12-09:
@peterbarker probably not VFR_HUD just because it's so different then all the other fields there. SYSTEM_TIME maybe? It's at least an associated time...

peterbarker, 2020-12-09:
@WickedShell I suggested VFR_HUD because it is something that you see a lot of people use in HUD-ish situations - for people who fly time rather than voltage on their batteries.

peterbarker, 2020-12-10:
Guess what I found?

```
    <message id="264" name="FLIGHT_INFORMATION">
      <description>Information about flight since last arming.</description>
      <field type="uint32_t" name="time_boot_ms" units="ms">Timestamp (time since system boot).</field>
      <field type="uint64_t" name="arming_time_utc" units="us">Timestamp at arming (time since UNIX epoch) in UTC, 0 for unknown</field>
      <field type="uint64_t" name="takeoff_time_utc" units="us">Timestamp at takeoff (time since UNIX epoch) in UTC, 0 for unknown</field>
      <field type="uint64_t" name="flight_uuid">Universally unique identifier (UUID) of flight, should correspond to name of log files</field>
    </message>
```

WickedShell, 2020-12-11:
Let's be honest that looks both good for that, and if we stuck the log file number in that UUID that'd be very useful (although not what it is really intended for)

tridge, 2020-12-14:
maybe flight_uuid should be MCU ID plus log number in bottom 16 bits?

peterbarker, 2020-12-15:
@hamishwillee so before I implement this message, should we fix the message definition?

Message definition clearly states microseconds (64 bits) for arm and takeoff time (assumed to be the same).

PX4 Firmware's implementation is populating this with millisecond values.

We can fix the message definition by changing the units to milliseconds....

This message is inordinately wasteful in terms of space.  At today's call we kind of figured it was a low-interval message anyway - but perhaps we should try again?  QGC doesn't support it.... but its not WIP and it has been in place since 2016...

@WickedShell it literally says that it should correspond to the log numbers ;-)  OK, so I'm going to skip over the "universally unique" bit...  FWIW, PX4 seem to populate this from a parameter....

hamishwillee, 2020-12-15:
@peterbarker @WickedShell Re the UUID PX4  does populate that from a parameter, which is iterated on every flight. Upshot, yes, it is not treated as a UUID. Perhaps it should be (or at least we should clarify for users they can't trust it to be).

I'm not sure what the right "fix" for the message is - I would have said for PX4 to populate with us since we can't know who else has implemented this, and because every other similar timestamp is us.

I'll raise issue in PX4 and get back to you.

peterbarker, 2020-12-16:
@hamishwillee FWIW the question of "why do we care about &micro;s precision for arm and takeoff time?" was raised.

hamishwillee, 2020-12-16:
> "why do we care about µs precision for arm and takeoff time?" was raised.

Good question. I suspect this is just a copy-pasted timestamp. On the other hand, what harm does it do to multiply by 1000? The real problem is we don't know who else is using this/impact of change. I've added it to mavlink call agenda.

hamishwillee, 2020-12-24:
@dagar modified PX4 to match the documented behaviour in MAVLink: https://github.com/PX4/PX4-Autopilot/issues/16393

peterbarker, 2020-12-24:
So my thoughts on `FLIGHT_INFORMATION` after half-doing an implementation:
 - uint64_t is too big?  microseconds too precise?
 - doesn't handle not having UTC at arming/takeoff (can be fixed in comments)
 - doesn't indicate whether you're still flying or armed so has to be combined with other messages to tell you useful information (and obviously you don't get a frame of data, meaning that we tell you when you armed to the microsecond but if you want to know  when you disarmed you'll have to wait for a `HEARTBEAT` (often 3Hz or 1Hz) and check your status...
 - No sensible way of filling in `UUID`, nor of relating it to logs (esp for dataflash-over-mavlink and "blackbox" logging, but also when we start to allow rolling logs over to new files in-flight
 - you may land and takeoff without disarming - corner case but the message won't handle that well

I'm sure there are other things that might be useful in this sort of log - total distance travelled, perhaps.