ArduPilot/ardupilot #15230: Arming Issue on Loiter in 4.1.0dev

NitinJSanket, 2020-09-03:
## Bug report
The Copter 4.1.0dev firmware does not arm in Loiter Mode. 
I'm using No GPS and optical flow with TFMini lidar. The same thing works in 4.0.4dev and lower. 
Arming in Althold and Stabilize still work and I can switch after arming in either of those modes to Loiter. In Loiter the arm error message says "No Position information".

**Version**
4.1.0dev

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X4

**Hardware type**
Holybro Kakute F7 AIO

**Logs**
N/A

muramura, 2020-09-04:
ArduPilot will not arming until the message "APM: EKF2 IMU0 is using GPS" or "APM: EKF2 IMU1 is using GPS" is output in LOITER mode, as the position cannot be determined.
In the case of STABILIZE and ALTHOLD, the position is not needed and therefore arming is done. If the above message notification is sent while arming, the mode is changed to LOITER mode.
Before this message notification, LOITER mode will not be used.

Please confirm this in the BIN file.

[image]

rmackay9, 2021-08-03:
I think I will close this issue because I think that optical flow is working in Copter-4.1.

@ NitinJSanket  if you're still looking at this but can't get optical flow working could you please report in the [Copter-4.1 support area](https://discuss.ardupilot.org/c/arducopter/copter-41/159) with a log and I'll dig into this.  thanks!

closed 2021-08-03 by rmackay9