ArduPilot/ardupilot #15233: CubeOrange sensor detection fails after soft reboot

WickedShell, 2020-09-03:
## Bug report

**Issue details**

After a soft reboot the CubeOrange cube can fail the board validation checks, I got an error of:
`Config error: Board Validation $CHECK_IMU2_PRESENT Failed`  as it was working before this, and works after a hard reboot it's clearly still working. @tridge has also seen this error before while soft rebooting.

Also relevant as soft reboots happen after firmware updates, which makes this far more likely for a user to encounter this, and be lead astray/file an RMA.

**Version**
bfc35db8ff7ca3b (Tridge would have seen this on an earlier version)

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
CubeOrange

IamPete1, 2021-07-26:
@bugobliterator Fix has been merged, so I guess that this can be closed, any update @WickedShell ?

WickedShell, 2021-07-26:
I haven't seen it recently, but I haven't been updating firmware on Orange boards recently either. @tridge probably is a better source of a cross check because he was encountering this before, and I imagine he's upgraded orange's more recently then I have.

IamPete1, 2022-09-10:
@WickedShell No new reports, I'd like to close.

WickedShell, 2022-10-14:
Closing, I haven't seen it again, and I've been doing a lot of this type of operation recently.

closed 2022-10-14 by WickedShell