ArduPilot/ardupilot #15250: AC3.2.1: Disabling AC_FENCE causes a build error

muramura, 2020-09-06:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

I want to test the AC 3.2.1 quad with APM.
I want to verify the operation of AC3.2.1 by making it minimally functional.
I have disabled AC_FENCE and I get a build error.
I have a video of the APM quadcopter flying in the world that is still being shown on video sites.
I think eliminating the build error would be good for ArduCopter users around the world.
I will now make some changes to the source.
Where do I advertise it?

Translated with www.DeepL.com/Translator (free version)

**Issue details**

[image]

**Version**

AC3.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QUAD

**Hardware type**

APM

**Logs**
_Please provide a link to any relevant logs that show the issue_

muramura, 2020-09-07:
I would add a change to enable and disable access to fence.pre_arm_check() in the AC_FENCE preprocessor.

Where do I go to promote these changes?

[image]

rmackay9, 2020-09-07:
We don't support the AC3.2.1 version anymore though so I don't think will accept any patches to the code or the wiki for this old version.

peterbarker, 2020-09-08:
@muramura -san  we haven't attempted to run software on this platform for a very, very long time now.

As a development team we do not have the resources or inclination to help out on any tweaks users might want to introduce newer features onto their working machines.

Just asking here has already cost time from several developers to evaluate - I've looked and thought about this one 3 times now, Randy at least once and probably several other devs in passing that didn't respond.

You're obviously welcome to try to backport features, but we've moved forward as a team and we'd prefer not to have to bounce these requests for assistance.

closed 2020-09-08 by peterbarker