ArduPilot/ardupilot #15252: Copter: The servo set value exceeds the maximum value.

muramura, 2020-09-06:
## Bug report

I typed the command "servo set 9 2000" in SITL.
The output of the MAVLINK inspector servo 9 is set to 2000.
I believe that when the set value is greater than the maximum value, it is the maximum value and when it is less than the minimum value, it is the minimum value.

**Issue details**

SERVO 9 OUTPUT VALUE is 2000:
[image]

SERVO 9 MAX 1900, MIN 1100
[image]

**Version**
ArduCopter V4.1.0-dev (65268ee3)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QUAD

**Hardware type**

SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

Hwurzburg, 2020-09-11:
I may be mistaken, but I believe the current behavior is intended....and a change could impact users experience...

rmackay9, 2020-09-11:
It's arguable either way but I probably agree with @muramura that we should enforce the limits when we receive a do-set-servo command.

muramura, 2020-09-12:
@Hwurzburg san, @rmackay9 san.  Thanks for the comment, I've PR'd it and will close it.

closed 2020-09-12 by muramura