ArduPilot/ardupilot #15278: SITL Simulation runs too fast in Ubuntu 20.04, Scaling speed has no impact

NDevDrone, 2020-09-08:
## Bug report

**Issue details**

Simulation is too fast, 'sim_speedup' scaling parameter has no impact when running SITL in Ubuntu 20.04 , WSL2

Similar as described to this post: https://discuss.ardupilot.org/t/simulation-run-too-fast/59115

**Version**
4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane, Copter, Plane

**Hardware type**
CubeBlack, Linux 

**Logs**
N/A

IamPete1, 2021-08-16:
Is this still a issue? I would guess out Ubuntu 20 support has improved. @khancyr Maybe you have a tested?

khancyr, 2021-08-16:
I haven't tested on wsl2, but it is working correctly on Ubuntu 20.04. So it is probably good to close this

closed 2021-08-16 by IamPete1