ArduPilot/ardupilot #15279: autotest: did not receive lane switch event

peterbarker, 2020-09-08:
## Bug report

**Issue details**

autotest fails when it didn't receive some expected text.  MAVProxy didn't get it either.  Perhaps there was no lane switch?

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
aetr

**Hardware type**
SITL

**Logs**

https://api.travis-ci.org/v3/job/725173988/log.txt

```
AT-0922.7: ---------- MAGNETOMETER: Change X-Axis Offset  ----------
AT-0922.7: -
AT-0922.7: Sending param_request_read for (SIM_MAG_OFS_X)
AT-0922.7: Drained 335 messages from mav (5101.114693/s)
AT-0922.8: get_parameter(SIM_MAG_OFS_X): PARAM_VALUE {param_id : SIM_MAG_OFS_X, param_value : 5.0, param_type : 9, param_count : 1212, param_index : 65535}
AT-0922.8: Setting SIM_MAG2_OFS_X to 255.000000
AT-0922.8: Sending param_request_read for (SIM_MAG2_OFS_X)
AT-0922.8: Drained 0 messages from mav (0.000000/s)
AT-0922.9: get_parameter(SIM_MAG2_OFS_X): PARAM_VALUE {param_id : SIM_MAG2_OFS_X, param_value : 5.0, param_type : 9, param_count : 1212, param_index : 65535}
AT-0922.9: Sending param_request_read for (SIM_MAG2_OFS_X)
APM: EKF3 lane switch 0

AT-0923.0: Received expected text: ekf3 lane switch 0
AT-0923.0: Received expected text: ekf3 lane switch 0
AT-0923.0: Drained 347 messages from mav (4056.352931/s)
AT-0923.0: get_parameter(SIM_MAG2_OFS_X): PARAM_VALUE {param_id : SIM_MAG2_OFS_X, param_value : 255.0, param_type : 9, param_count : 1212, param_index : 65535}
AT-0923.0: Waiting for text : ekf3 lane switch
AT-0928.2: Caught exception: Traceback (most recent call last):
  File "/home/travis/build/ArduPilot/ardupilot/Tools/autotest/arduplane.py", line 1930, in ekf_lane_switch
    self.wait_statustext(text="EKF3 lane switch", timeout=30)
  File "/home/travis/build/ArduPilot/ardupilot/Tools/autotest/common.py", line 4088, in wait_statustext
    text.lower())
AutoTestTimeoutException: Failed to receive text: ekf3 lane switch1
```

```
AT-0931.5: Uploading 21 logs to https://firmware.ardupilot.org/CI-Logs/2020-09-08-14-10
```

@harshitsankhla

harshitsankhla, 2020-09-09:
@peterbarker 
a lane switch does happen. It seems to be some issues with speeding the test up.
its waiting for the lane switch status text after the lane switch has happened so the status text is gone
to trigger a lane switch we change a parameter, in this case - SIM_MAG2_OFS_X
first get the parameter, then set it to the new value, then again get the parameter to see if it has actually changed, then wait for lane switch status text which, for this test to work, should occur now, but the status text pops up just before the changed parameter is read back

also there is a logical error in the test that I just noticed, the old parameter value should be SIM_MAG2_OFS_X, and not SIM_MAG_OFS_X
 
> https://api.travis-ci.org/v3/job/725173988/log.txt
this shows passed the test

peterbarker, 2020-09-10:
On Tue, 8 Sep 2020, Harshit Kumar Sankhla wrote:

> a lane switch does happen. It seems to be some issues with speeding the test up.
> its waiting for the lane switch status text after the lane switch has happened so the status text is gone

OK, in that case we should be taking a collection of statustexts and 
checking it afterwards.  An example of where we do that is here:

https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arduplane.py#L920

> also there is a logical error in the test that I just noticed, the old parameter value should be SIM_MAG2_OFS_X, and not SIM_MAG_OFS_X

Could you add a patch for that?

harshitsankhla, 2020-09-10:
> Could you add a patch for that?

#15303

PR #15303 (2020-09-11): autotest: bug fixes for ekf lane switch test
the EKF lane switch test in Plane would occasionally fail, mostly because the lane switch status text would sometimes pop up after the parameter change to cause the lane switch but before we actually start looking for it, this PR fixes that bug by collecting status text messages from the beginning of the subtest and look for the lane switch one in context when required. This ensures we do not miss it.

also I've checked this with #15281 and its working fine with the speedups (the aforementioned PR is failing the ekf lane switch test on CI)

fixes #1 [... cut]
closed 2020-09-11 by peterbarker