ArduPilot/ardupilot #15296: Copter/Rover: follow mode should have choice of abort behavior

patrickpoirier51, 2020-09-10:
When in follow mode, if we loose connection with leader (_SYSID)  for more than 10 Seconds, it will  assign the leader ID to "0" that mean follow anyone in the DIST_MAX

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Follow/AP_Follow.cpp#L27
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Follow/AP_Follow.cpp#L270
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Follow/AP_Follow.cpp#L318

This behavior is quite unpredictive and can lead to problems.
I think it would be much safer if we could set the Behavior and TimeOut  as parameter that include Land, RTL, Smart RTL or else

IamPete1, 2020-09-10:
https://github.com/IamPete1/ardupilot/tree/follow_timeout

Completely untested, RTL only, if there is consensus that his is a good idea I can finish it off.

rmackay9, 2020-09-11:
I think it will only reset FOLL_SYSID back to zero if it was zero when entering follow mode.  The idea was that if the user was happy not to specify what to follow, then after the timeout they would be equally happy to not have it specified.  The alternative is that once it locks onto a target it never changes the sysid again until rebooted.

If @IamPete1 or anyone else wants to implement this I think good names for the parameters would be:

- FOLL_TIMEOUT <-- parameter to specify what the timeout should be in seconds.  0 should disable the timeout although this is equivalent to setting the parm below to 0.
- FOLL_TIMEOUT_ACT <-- parameter to specify the action that should be taken.  valid entries could be, 0:Wait, 1:RTL, 2:SmartRTL or RTL, 3:SmartRTL or Land, 4: Land.  Note that these values should be vehicle specific because Rover would have slightly different options.  For Rover Hold would probably replace Land and a Loiter option might be required... or perhaps "Hold or Loiter" could be the option and rovers would Hold while boats would Loiter.

patrickpoirier51, 2020-09-14:
@IamPete1 

Peter,  after succesfull test ,  I think that suggestions from @rmackay9  would make  sense .
Standing by for further testing :-)

Thanks to both