ArduPilot/ardupilot #15298: Hereflow + TF Mini don't work in same time...

multinnov, 2020-09-10:
Hello,

I have identified a rather particular problem...
I have equipped a Hereflow and a TF mini (for avoiding obstacles upwards so as not to hit the ceilings) with a Mrobotics Zero F7. 

We worked rather well, I had my LIDAR altitude measurement up to 2 meters, and the LIDAR upwards to stop the drone at 2 meters from the ceiling. I had to fly 4-5 times without too many problems (a priori).

However I noticed that systematically, that the LIDAR upwards is "cut/non functional" as soon as the Lidar downwards (hereflow is activated). As soon as I deactivate the LIDAR with "RNGFND2_TYPE"=None, the top LIDAR (RNGFND1) works.

As this, there was a priority on the Hereflow (CAN). The SERIAL is set up correctly.

In fact all works "sometimes", but there it becomes very rare that it works, or even more at all... So I don't really understand... On another drone, with the same equipment, I have the same symptoms...

The code that worked, but cuts the LIDAR from above.

Surprisingly:
When the top LIDAR is activated alone, in the "status" tab, I have a measurement on rangefinder1, but if I activate "PRX_TYPE=4", then this one is no longer functional, rangefinder goes to 0.

When I activate the lower LIDAR and deactivate the upper one, then in "status" to "sonarrange" I have many coherent values (nothing to "rangefinder2").

When I then activate the top LIDAR (in addition to the bottom one), then no value at "rangefinder1", but I have the same value at "rangefinder2" as "sonnarange".

Thank you

Regards

multinnov, 2020-09-11:
`ACRO_BAL_PITCH,1
ACRO_BAL_ROLL,1
ACRO_RP_EXPO,-0.2
ACRO_RP_P,4.5
ACRO_THR_MID,0
ACRO_TRAINER,2
ACRO_Y_EXPO,0.2
ACRO_YAW_P,4.5
ADSB_ENABLE,0
AHRS_COMP_BETA,0.1
AHRS_CUSTOM_PIT,0
AHRS_CUSTOM_ROLL,0
AHRS_CUSTOM_YAW,0
AHRS_EKF_TYPE,3
AHRS_GPS_GAIN,1
AHRS_GPS_MINSATS,0
AHRS_GPS_USE,0
AHRS_ORIENTATION,2
AHRS_RP_P,0.2
AHRS_TRIM_X,0.005
AHRS_TRIM_Y,-0.025
AHRS_TRIM_Z,0
AHRS_WIND_MAX,0
AHRS_YAW_P,0.2
ANGLE_MAX,1200
ARMING_ACCTHRESH,0.75
ARMING_CHECK,0
ARMING_MIS_ITEMS,0
ARMING_RUDDER,2
ATC_ACCEL_P_MAX,110000
ATC_ACCEL_R_MAX,110000
ATC_ACCEL_Y_MAX,18000
ATC_ANG_LIM_TC,1
ATC_ANG_PIT_P,4.5
ATC_ANG_RLL_P,4.5
ATC_ANG_YAW_P,4.5
ATC_ANGLE_BOOST,1
ATC_INPUT_TC,0.15
ATC_RAT_PIT_D,0.0036
ATC_RAT_PIT_FF,0
ATC_RAT_PIT_FLTD,20
ATC_RAT_PIT_FLTE,0
ATC_RAT_PIT_FLTT,20
ATC_RAT_PIT_I,0.135
ATC_RAT_PIT_IMAX,0.5
ATC_RAT_PIT_P,0.18
ATC_RAT_RLL_D,0.0036
ATC_RAT_RLL_FF,0
ATC_RAT_RLL_FLTD,20
ATC_RAT_RLL_FLTE,0
ATC_RAT_RLL_FLTT,20
ATC_RAT_RLL_I,0.135
ATC_RAT_RLL_IMAX,0.5
ATC_RAT_RLL_P,0.18
ATC_RAT_YAW_D,0
ATC_RAT_YAW_FF,0
ATC_RAT_YAW_FLTD,0
ATC_RAT_YAW_FLTE,2.5
ATC_RAT_YAW_FLTT,20
ATC_RAT_YAW_I,0.8
ATC_RAT_YAW_IMAX,0.5
ATC_RAT_YAW_P,0.6
ATC_RATE_FF_ENAB,1
ATC_RATE_P_MAX,0
ATC_RATE_R_MAX,0
ATC_RATE_Y_MAX,0
ATC_SLEW_YAW,6000
ATC_THR_MIX_MAN,0.1
ATC_THR_MIX_MAX,0.5
ATC_THR_MIX_MIN,0.1
[... 953 lines cut ...]
TERRAIN_ENABLE,1
TERRAIN_FOLLOW,0
TERRAIN_SPACING,100
THR_DZ,100
THROW_MOT_START,0
THROW_NEXTMODE,18
THROW_TYPE,0
TUNE,0
TUNE_MAX,0
TUNE_MIN,0
VISO_ORIENT,0
VISO_POS_X,0
VISO_POS_Y,0
VISO_POS_Z,0
VISO_TYPE,0
WP_NAVALT_MIN,0
WP_YAW_BEHAVIOR,2
WPNAV_ACCEL,100
WPNAV_ACCEL_Z,100
WPNAV_RADIUS,200
WPNAV_RFND_USE,1
WPNAV_SPEED,500
WPNAV_SPEED_DN,150
WPNAV_SPEED_UP,250
`

KipK, 2021-01-12:
Same issue here. 2 UART TF-mini and one CAN Hereflow. 
I still can use the Hereflow optical flow while tf-mini are operationals, but activating the embedded lidar will make the TF-Mini rangefinders non functionnal.